TrampolineRTOS/trampoline #6: Commit https://github.com/TrampolineRTOS/trampoline/commit/1c6896ba877fb77edeb16e181cb5ad941848eef6#diff-226ddbef4ee95a20247f543d5afccafe breaks posix

jlbirccyn, 2015-11-16:
Commit https://github.com/TrampolineRTOS/trampoline/commit/1c6896ba877fb77edeb16e181cb5ad941848eef6#diff-226ddbef4ee95a20247f543d5afccafe in root.goilTemplate breaks posix port. Posix port does not use the `SOURCE` attribute for IRQ. Computation based on the target specific OIL attributes should be done in a target specific template

DavidGarriou, 2015-11-16:
Modification of files:
```goil/templates/root.goilTemplate```

Add files:
```goil/templates/code/cortex/target_specific.goilTemplate```

Commit
[e1cb57e]

with message:
```[issue][#6] Fix for root.goilTemplate regression```

DavidGarriou, 2015-11-16:
New commit because the previous did break cortex port:

Modification of files:
```
goil/templates/code/cortex/armv7/counter_call.goilTemplate
goil/templates/code/cortex/armv7/interrupt_table.goilTemplate
goil/templates/code/cortex/armv7/tpl_interrupt_vectors.goilTemplate
goil/templates/code/cortex/armv7/tpl_primary_irq.goilTemplate 
goil/templates/root.goilTemplate
```

Deletion of:
```
goil/templates/code/cortex/target_specific.goilTemplate
```

Commit
```
92b1e9d
```
with message:
```
[cortex][issue][#6] Fix of cortex port for IRQ structures generation
```

closed 2015-11-16 by DavidGarriou