TrampolineRTOS/trampoline #12: Optimize assembly code

DavidGarriou, 2015-12-08:
Suppress from machine system calls and irq handling:
``
push {lr}
...
pop {lr}
```

Optimize use of registers from machine system calls and irq handling:
sp register is saved into r5 and then r5 is used all along the handlers.
This useless; Don't save sp, and use it without modifying it.
Study whether or not r4 is useful. We shall use either r0, r1, r2 at the very end of handler because 
they will be restored on return.

jlbirccyn, 2016-01-15:
Fixed in commit https://github.com/TrampolineRTOS/trampoline/commit/eb8f123fb62580f8190571a6526b94c7155a652c

DavidGarriou, 2016-01-18:
I need to enhance use of register r4 and r5 before closing the ticket

closed 2019-10-20 by jlbirccyn