TrampolineRTOS/trampoline #13: Cross Compiling on Posix MACHINE

urnenfeld, 2016-01-05:
I have a yocto recipe which would allow to deploy Trampoline in Linux Embedded systems.
The task was a success but there was an issue which was not solved as clean as it could be.

The makefile generated by goil contains the following lines:
---
#############################################################################
# COMPILER CONFIGURATION
#############################################################################
[...]
CC = gcc
[...]
LD = gcc
[...]
AS = gcc
---

These 3 statements override the environment variables set by yocto to ensure proper cross compiling. The "solution" was to remove such lines from the makefile.

Maybe the generation could be parametrized or make a weak assignment of these variables?

jlbirccyn, 2016-01-05:
Hello. Yes we will fix that

sefau, 2016-01-05:
This has been implemented in branch ticket13. It will be merged in the master branch after some testing.

sefau, 2016-01-06:
Update has been merged.
If the compiler, assembler and linker are not set in the project OIL file, the value given by the environment variable CC, AS and LD will be used over the default value (gcc).

closed 2016-01-06 by sefau

urnenfeld, 2016-01-07:
Tested from my side. The recipe has been updated as well to fetch the new github repository & works as expected. Thanks

jlbirccyn, 2016-01-07:
You are welcome :-)