TrampolineRTOS/trampoline #14: [templates] [goil] Kernel.oil not taken into account in python build system

LMG, 2016-01-07:
When adding a category in goil/templates/config/kernel.oil , the python build system fails because it can not find the corresponding header files.

jlbirccyn, 2016-01-08:
Is there a way to know if the control module is used in the OIL file ? (custom attribute of tasks ?)

LMG, 2016-01-09:
Not as of now, but it could be added (and used for other purposes later on like locating the model description file).

LMG, 2016-01-09:
wrong button :)

jlbirccyn, 2016-01-11:
Ok, let's add a boolean key in the OS object to handle this case. Something like
``` 
CONTROL = TRUE;
```

LMG, 2016-01-15:
The pull request above offers a solution, explained in [kernel module](https://github.com/TrampolineRTOS/trampoline/wiki/Kernel-modules)

PR #19 (2016-01-15): [build] Support for kernel modules
Base to fix #14
closed 2016-01-15 by LMG