TrampolineRTOS/trampoline #16: compiler on avr/arduino port

mbriday, 2016-01-08:
The examples in the AVR/arduino port do not compile. Default compiler is the one of the host

sefau, 2016-01-08:
This is a side effect of the fix introduced in commit [64294eb] (https://github.com/TrampolineRTOS/trampoline/commit/64294eb5a79ad8cd6ac0309247aa66360e41c700). Now the compiler, assembler and linker are set according to the following rules:
1. take the values defined in the project OIL file
2. if no value is given in the project OIL file, take the values defined in the CC, AS and LD environment variables
3. if the environment variables are not set, take the default values (gcc).

To fix the problem w/ the AVR/arduino examples, I guess that the correct values should be set in the project OIL files. I do not have a cross-compilation chain for this target to test this solution. Please let me know if it works.

mbriday, 2016-01-08:
Fixed in commit [c9c26e8](https://github.com/TrampolineRTOS/trampoline/commit/c9c26e831a3a56d35f86d74a907240d12c94bff9). I removed the default compiler in the goil templates. It is now explicitely defined for each example.

closed 2016-01-08 by mbriday