TrampolineRTOS/trampoline #33: python build system does not run under python 3.x

mbriday, 2016-04-28:
The python build system is limited to python 2.x

Snakeuh, 2016-05-04:
Made it "work" with python 3.* with this patch.

[0001-enhancement-Fixes-33.patch.txt](https://github.com/TrampolineRTOS/trampoline/files/248988/0001-enhancement-Fixes-33.patch.txt)

By "work", i mean it compiles and produces an usable binary, however terminationSemaphore.acquire() doesn't see that a job has finished (so that a specific file has finished its compilation) so I had to manually stop and launch again the scripts each time a sequence of files had finished their compilation. It may be due to this computer's python environment though (ArchLinux python 3.5.1).

jlbirccyn, 2016-06-02:
Updated to the latest version of the script which is now compatible with Python 2 _and_ Python 3. If someone could check it is ok with Python 3 (I do not have Python 3 installed on my computer).

Snakeuh, 2016-06-02:
Commit 7b22091 : Copied the modifications of makefile.py plus a minor change in generic_galgas_makefile.py support python3 goil's compilation. Minor change too for the generation of build.py. Everything seems to work fine

jlbirccyn, 2016-06-03:
Do we consider this issue is fixed ?

jlbirccyn, 2016-06-03:
FIxed in 2839a7a059914501f4bdf47ff9f7751f70a8514e

closed 2016-06-03 by jlbirccyn