TrampolineRTOS/trampoline #37: STM8 port

spth, 2016-06-24:
Would it be possible to port this OS to the STM8? What would have to be done?

Philipp

DavidGarriou, 2016-06-24:
Hi,

yes it is possible to port Trampoline on a STM8.
The main work has to be done on cpu specific features : 
- startup process
- switch context
Then one must connect a timer to the OS tick timer.
At last one must modify templates files related to the CPU for : 
- interrupt generation,
- service call.

There is a documentation for porting trampoline.
```
documentation/manual/main.mdf
```

Would you plan to port it yourself ?

David

jlbirccyn, 2016-06-24:
David won by 10 minutes :)

STM8 seems to be a quite standard 8 bits accumulator ISA. So yes, it can be ported. I read the stack has to be located in page 0 (64kB from 0 to 0xFFFF) so all the stacks of the tasks/ISR2 and the kernel stack have to be less than 64kB minus the vectors.

Just updated chapter 13 of the manual about porting. It is a good start and if you have any question, feel free to ask.

spth, 2016-06-24:
I might attempt a port.

The STM8 is rather small. The biggest devices have 6 KB of RAM. The few devices that use the address space above 0xffff have flash there.

I have previously created an STM8 port for the Small Device C Compiler for the RTOS Atomthreads. It was quite easy. But Atomthreads already had STM8 ports for other compilers. Also a knowledge of C, Makefiles and the target architecture was sufficient there (no servie calls, GOIL or weird macros like FUNC, OS_CODE, etc).
Porting Trampoline probably would be much harder.

Philipp

closed 2016-10-02 by jlbirccyn