TrampolineRTOS/trampoline #44: [Arduino Uno] 'CallTerminateISR2' undeclared

guillaume-chs, 2016-10-19:
### Hi guys !
While trying to setup a **lightweight docker container for compiling & flashing or Arduino Uno** specifically, I've encountered this issue after pulling **TrampolineRTOS last version** :

`./make.py`
../../../../machines/avr/tpl_machine.c: In function 'tpl_init_context':
../../../../machines/avr/tpl_machine.c:63:18: error: **'CallTerminateISR2' undeclared** (first use in this function)
   AVR_PUSH((u16)(CallTerminateISR2) & 0xFF)

### This fixed the issue
https://github.com/guillaume-chs/trampoline-for-docker-trampuino/commit/8a6635f779aef2401b5f08812d652265dc60d2a9

By the way, I hope very much that you are OK with my work, I solely intend to create a docker container to be run as a compiler, for the purpose of the "Systems & Networks" course at the "Ecole des Mines de Nantes".
Docker container repo : https://github.com/guillaume-chs/trampuino
My fork of TrampolineRTOS : https://github.com/guillaume-chs/trampoline-for-docker-trampuino

Cheers,
Guillaume !

Snakeuh, 2016-10-19:
Hello !

You're absolutly right.  This bug is due to commit 173b06aba4c0c97fd3af3d137e94ee9585cb4474 : adding the flag HIDDEN to the CallTerminateISR2 syscall have as well removed the function declaration from the generated files to avoid the user to call directly this function (same as CallTerminateTask).

In fact, CallTerminateISR2 has to be declared as external in the same file in which the address of the function is used as you did. Other mistake just below : we're pushing TerminateTask instead of CallTerminateTask in the stack.

Thanks for pointing out this bug !
Kamel

closed 2016-10-20 by Snakeuh

jlbirccyn, 2016-10-20:
Thanks Guillaume for your contribution. 

By the way your work is very welcome ;-)

Best regards

commit e5b320db (2016-10-20): [fix] Fixes #44