TrampolineRTOS/trampoline #45: [GOIL] Is there a build-in way to add a library without updating goil templates ?

Dnomyar, 2016-11-06:
I would like to be able to use a library (U8glib) in a simple project. 
So I have imported this lib in `lib/U8glib` and included it `#include "../lib/U8glib/U8glib.h"` (I have a `src/screenJoyStick.h`).

But, I get some errors :
```
...
[100%] Linking trampuinoScreenJoyStick
build/src/screenJoyStick.cpp.o: In function `U8GLIB::write(unsigned char)':
screenJoyStick.cpp:(.text._ZN6U8GLIB5writeEh[U8GLIB::write(unsigned char)]+0xe): undefined reference to `u8g_DrawGlyph'
build/src/screenJoyStick.cpp.o: In function `periodicTask_function':
screenJoyStick.cpp:(.text.periodicTask_function+0x12): undefined reference to `u8g_Begin'
screenJoyStick.cpp:(.text.periodicTask_function+0x1a): undefined reference to `u8g_FirstPage'
build/src/screenJoyStick.cpp.o: In function `_GLOBAL__sub_I_u8g':
screenJoyStick.cpp:(.text.startup._GLOBAL__sub_I_u8g+0x28): undefined reference to `u8g_dev_st7565_nhd_c12864_sw_spi'
screenJoyStick.cpp:(.text.startup._GLOBAL__sub_I_u8g+0x2a): undefined reference to `u8g_dev_st7565_nhd_c12864_sw_spi'
screenJoyStick.cpp:(.text.startup._GLOBAL__sub_I_u8g+0x2e): undefined reference to `U8GLIB::initSPI(_u8g_dev_t*, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)'
collect2: error: ld returned 1 exit status
Return code: 1
```

I guess this errors are linking errors. The lib should not be included in compilation and linking process.
@guillaume-chs has fixed this here https://github.com/guillaume-chs/trampoline-for-docker-trampuino/commit/06b957142723f81e6c2605ea42b32c10e696dfd9
This works but, update goil template to add a library support is not really extensible.
**Do you have a solution to specify `lib` directory in goil configuration file ?**

I have the same problem if I split my C++ code in classes : 
```
Nothing to make.
[  5%] Compiling /path/to/trampoline/os/tpl_os_kernel.c
[ 11%] Compiling /path/to/trampoline/os/tpl_os_timeobj_kernel.c
[ 17%] Compiling /path/to/trampoline/os/tpl_os_action.c
[ 23%] Compiling /path/to/trampoline/os/tpl_os_error.c
[ 29%] Compiling /path/to/trampoline/os/tpl_os_os_kernel.c
[ 35%] Compiling /path/to/trampoline/os/tpl_os_os.c
[ 41%] Compiling /path/to/trampoline/os/tpl_os_interrupt_kernel.c
[ 47%] Compiling /path/to/trampoline/os/tpl_os_task_kernel.c
[ 52%] Compiling /path/to/trampoline/os/tpl_os_resource_kernel.c
[ 58%] Compiling /path/to/trampoline/os/tpl_os_alarm_kernel.c
[ 64%] Compiling src/screenJoyStick.cpp
[ 70%] Compiling screenJoyStick/tpl_app_config.c
[ 76%] Compiling screenJoyStick/tpl_dispatch_table.c
[ 82%] Compiling screenJoyStick/tpl_invoque.S
[ 88%] Compiling /path/to/trampoline/machines/avr/tpl_machine.c
[ 94%] Compiling /path/to/trampoline/machines/avr/arduino/main.cpp
[100%] Linking trampuinoScreenJoyStick
build/src/screenJoyStick.cpp.o: In function `setup':
screenJoyStick.cpp:(.text.setup+0x4): undefined reference to `JoyStick::getJoyStickPosition()'
build/src/screenJoyStick.cpp.o: In function `_GLOBAL__sub_I_js':
screenJoyStick.cpp:(.text.startup._GLOBAL__sub_I_js+0x8): undefined reference to `JoyStick::JoyStick(unsigned int)'
collect2: error: ld returned 1 exit status
Return code: 1
```

To generalize my question : **do you have a solution to specify others C++ scripts in goil configuration file ?**

Thanks for your great work!

Snakeuh, 2016-11-07:
Hello !

I think you'd want to use the "LIBRARY" oil definition.
Check for the oil files located in the trampoline/tests/machines directory for some examples.
Be careful, the library's relative paths are from the "trampoline/machines" directory.

In your case, something like this should do : 
```
CPU myCPU
{
   OS myOS {
      BUILD = TRUE {
         LIBRARY = myglib;
      };
   };
   LIBRARY myglib
   {
       PATH = "path/to/U8glib";
       CPPHEADER = "U8glib.h";
       CPPFILE = "U8glib.cpp";
    };
};
```

See you,

Kamel Hacene

jlbirccyn, 2016-11-07:
Hello,

Thanks for you interest. Another answer: If you want to add an already compiled library to the link phase, you can use the LDFLAGS attribute in the OIL file. This attribute is a sub attribute of the BUILD attribute (when true) in OS object. For example:

```
  OS config {
    STATUS = STANDARD;
    ARDUINO = TRUE {
      SERIAL = TRUE; //use the serial interface 
    };
    BUILD = TRUE {
      TRAMPOLINE_BASE_PATH = "../../../..";
      APP_NAME = "trampuinoBlink";
      APP_SRC = "blink.cpp";
      CPPCOMPILER = "avr-g++";
      COMPILER = "avr-gcc";
      LINKER = "avr-gcc";
      ASSEMBLER = "avr-gcc";
      COPIER = "avr-objcopy";
      SYSTEM = PYTHON;
      LDFLAGS = "-L/path/to/my/library -lmyLib";
    };
    SYSTEM_CALL = TRUE;
  };
```

Dnomyar, 2016-11-07:
Hello !
Thanks for your quick answers !

I have set up the solution you gave to me.

Here is the goil configuration file : 
```
OIL_VERSION = "2.5" : "test" ;

CPU test {    
  OS config {
    STATUS = STANDARD;
    ARDUINO = TRUE {
      SERIAL = TRUE; //use the serial interface 
      PORT = "/dev/cu.usbmodem1421"; //serial port (flash)
    };
    
    BUILD = TRUE {
      TRAMPOLINE_BASE_PATH = "/path/to/trampoline";
      APP_NAME = "trampuinoScreenJoyStick";
      APP_SRC = "src/screenJoyStick.cpp";
      CPPCOMPILER = "avr-g++";
      COMPILER = "avr-gcc";
      LINKER = "avr-gcc";
      ASSEMBLER = "avr-gcc";
      COPIER = "avr-objcopy";
      SYSTEM = PYTHON;
      LIBRARY = u8gb;
    };
    SYSTEM_CALL = TRUE;
  };

  LIBRARY u8gb {
     PATH = "../lib/U8glib";
  };

  APPMODE stdAppmode {};
  
  ALARM periodicAl {
    COUNTER= SystemCounter;
    ACTION = ACTIVATETASK {
      TASK = periodicTask;
    };
    AUTOSTART = TRUE {
      //ALARMTIME = 25;
      //CYCLETIME = 25;
      ALARMTIME = 25;
      CYCLETIME = 25;
      APPMODE = stdAppmode;
    };
  };

  TASK periodicTask {
    PRIORITY = 20;
    //AUTOSTART = FALSE;
    AUTOSTART = TRUE {APPMODE=stdAppmode;};
    ACTIVATION = 1;
    SCHEDULE = FULL;
    STACKSIZE = 256;
  };

};
```
When I run `goil` and `make.py`, I have the following logs : 
```
Compiling
Replaced 'screenJoyStick/tpl_dispatch_table.c'.
Replaced 'screenJoyStick/tpl_invoque.S'.
Replaced 'screenJoyStick/tpl_invoque_isr.S'.
Replaced 'screenJoyStick/tpl_os.h'.
[... 405 lines cut ...]
make.runGoal(maxParallelJobs, maxParallelJobs == 1)

#----------------------------------------------------------------------
#--- post commands
#----------------------------------------------------------------------
if make.errorCount() == 0:
  if askedGoal == "flash" :
    commandLine = "avrdude -c arduino -p m328p "
    if os.environ.get('AVRDUDE_PORT') != None:
      commandLine += "-P" + os.environ['AVRDUDE_PORT'] + " "
    else:
      print("Cannot flash, environment variable AVRDUDE_PORT is not defined")
      exit(1)
    commandLine += "-Uflash:w:"
    commandLine += "trampuinoScreenJoyStick.hex "
    if makefile.find_executable("avrdude") != None:
      print(commandLine)
      subprocess.call(commandLine, shell=True)
    else:
      print("Command 'avrdude' not in PATH")

#----------------------------------------------------------------------
#-- End of build.py
#----------------------------------------------------------------------
```

Snakeuh, 2016-11-07:
The LIBRARY's PATH property defines the path to your library's directory. The past just has to be relative from your Trampoline's machines directory.

I've downloaded your library and managed to compile it with an example without problem. I don't have the full path to your lib directory so i can't give you what you need to write in that field but here's exactly what i did : 

Let's say the trampoline directory is $TRAMPOLINE, and i've put the library in $TRAMPOLINE/examples/avr/arduinoUno/myExample/U8glib/
```
CPU mycpu {
  OS myos {
    BUILD {
      LIBRARY = glib;
    };
  };

  LIBRARY glib
  {
    PATH = "../examples/avr/arduinoUno/myExample/U8glib";
    CPPFILE = "U8glib.cpp";
    CPPHEADER = "U8glib.h";
  };
};
```

PATH : Tells Goil where to find the Library files.
CPPFILE : List of files to compile with your code.
CPPHEADER : List of headers to include in your code.

Also, I see that there is an "utility" directory in U8glib, so maybe you will have to add some of these files in another library like that : 
```
CPU mycpu {
  OS myos {
    BUILD {
      LIBRARY = glib;
      LIBRARY = glib_utils;
    };
  };

  LIBRARY glib
  {
    PATH = "../examples/avr/arduinoUno/myExample/U8glib";
    CPPFILE = "U8glib.cpp";
    CPPHEADER = "U8glib.h";
  };

  LIBRARY glib_utils
  {
    PATH = "../examples/avr/arduinoUno/myExample/U8glib/utility";
    CFILE = "file1.c";
    CFILE = "file2.c";
    (...)
  };
};
```

See you

jlbirccyn, 2016-11-08:
Hello

Seems your TRAMPOLINE_BASE_PATH attribute is not set properly. /path/to/trampoline is a pseudo path intended to make the user understand it has to be replaced with your actual path to the trampoline directory. 

Best regards

Dnomyar, 2016-11-08:
Thanks for your very detailed answers.
It helped me a lot to fix my issue.
All the best for your project.
Best regards

closed 2016-11-08 by Dnomyar