TrampolineRTOS/trampoline #46: Problem with STM32

besnarval, 2016-12-11:
Bonjour,

Je travaille actuellement sur Trampoline avec la cible STM32 F407VG dans le cadre d'un projet à l'ESEO. Je suis confronté à un problème que je n'arrive pas à résoudre depuis plus de 3 ou 4 semaines.
Je travaille sur un code de mise en oeuvre de l'UART avec transmission et réception en interruption. Après avoir testé le code en baremetal, on s'est rendu compte que la STM32 arrive à tenir un baudrate de 115200 sans souci. Quand je teste ce code avec Trampoline, j'ai un HardFault qui se produit et la carte crashe.
Je pense qu'il s'agit d'un problème de temps réel (mavaise préemption ou quelque chose du genre) car le code marche en mode pas à pas. J'obtiens également des comportements différents suivant le nombre de caractères que j'envoie : 6 caractères le code marche, 7 caractères le code ne marche pas, une trentaine de caractères ca marche. Il a quelque chose de louche qui se produit comme si l'arrivée d'un évènement à un moment très précis faisait crasher la carte.
Voici le backtrace que  j'obtiens quand la carte crashe :
#0  HardFault_Handler ()
    at /home/valentin/I3/PFE/trampoline/machines/cortex/armv7em/stm32f407/handlers_stm32f4xx.c:80
#1  <signal handler called>
#2  tpl_save_context ()
    at /home/valentin/I3/PFE/trampoline/machines/cortex/tpl_ctx_switch.S:85
#3  0x08001ade in tpl_primary_irq_handler_USART2_IRQ ()
    at uart/tpl_primary_irq.S:407
#4  <signal handler called>
Backtrace stopped: Cannot access memory at address 0x20020018

Ce qu'on peut voir c'est que c'est un changement de contexte suite à l'interruption UART qui déclenche le HardFault. Le backtrace n'arrive pas à accéder à l'adresse 0x20020018. En effet, la RAM va de 0x20000000 à 0x20020000 donc il n'y a pas de case mémoire à l'adresse 0x20020018. 

Le problème vient peut-être d'une mauvaise utilisation de Trampoline de ma part. J'ai protéger les sections critiques par DisableAllInterrupts() / EnableAllInterrupts() (sauf celle de l'ISR qui ne peut pas être préemptée). Est ce bien ces fonctions que je dois utiliser pour protéger les sections critiques ? 
Y a t'il des priorités spéciales à mettre pour les interruptions NVIC ? J'ai lu qu'il avait des problèmes de priorités entre la STM32 et l'OS avec d'autres RTOS différent de Trampoline. Le problème pourrait peut-être venir de là.

Voici mon code avec l'executable et le .map après compilation
[uart_test_4.zip](https://github.com/TrampolineRTOS/trampoline/files/644350/uart_test_4.zip)

Valentin BESNARD

jlbirccyn, 2016-12-12:
Hello Valentin

A quick answer after examining your code. It may or may not be the right answer though.

In Trampoline on Cortex-M, the supervisor call exception (SVC) priority is set to 6. According to what I see in your code, you do not change the priority of the UART interrupt. So by default it is set to 0 (David, if I say something wrong correct it)  which is a higher priority than the priority of the SVC. So the UART can interrupt the kernel. The UART ISR is an ISR2 so it ends by calling CallTerminateISR2 service. If the kernel has been interrupted, this is a direct call to tpl_call_terminate_isr2_service function and a race condition occurs on kernel structures (or if it happens just when the reentrancy counter is updated, it can be worse).

Add a 

```
  NVIC_SetPriority(USART2_IRQn, 6);
```

In your initialization code.

In the future these init will be in generated code but for now, we have to do some inits by hand.

If it does not work, I will have a little bit time to look at it tomorrow.

Best regards

besnarval, 2016-12-12:
Hello,

That seems better with this priority. The STM32 doesn't crash but the protection of critical sections with DisableAllInterrupts() / EnableAllInterrupts doesn't work. My "index_write" takes the good value at the beginning but sometimes this index was not incremented. 
I run my code and I send this original string "abcdefgh".
Approximately after 10 seconds, the string is transformed into "**e**bcdefgh" because the index was not incremented. I had the same behaviour with an ISR1 yesterday.

What is the best to protect critical sections of code ?

Best regards

Valentin

DavidGarriou, 2016-12-12:
Hi,

Glad to hear that your driver works.
The DisableAllInterrups and EnableAllInterrupts functions actually do not work for this port.
This is a work in progress, i keep you in touch as soon as i have finished.
You should perhaps use the resources for critical access to variable.
I did have a very light eye on your code and it appears that you wrote a lot of stuff into 
acknowledgement functions for IRQs. And you do nothing but toggling a led into the body of ISR2.
It appears to me that you need ISR1 and not ISR2.

David

David Garriou
+33 613 504 246

jlbirccyn, 2016-12-12:
David is right but while waiting for working DisableAllInterrups and EnableAllInterrupts services, you can continue to use an ISR2 and use a resource you attribute to the task and to the ISR2 (adding a RESOURCE=<the name of your resource>; in both task and ISR objects. By using GetResource and ReleaseResource services, you enter and leave the critical section.

besnarval, 2016-12-12:
Ok thanks you for your anwers. I will try to protect critical sections with ressource. I keep you in touch when I will have test it.

@DavidGarriou : Yes, I do a lot of stuff in my acknowledgement function for IRQ because I need to know which kind of interrupt (reception or transmission) triggers the interrupt.
My problem is that if I clear the flag in the acknowledgement function, then in the ISR I can't know if the interruption was triggered for a reception or a transmission and I can't do the correct treatment.
Moreover, if I don't clear the flag in the acknowledgement function, the ISR was not called after the acknowledgement function. Is that a normal behaviour ?

Valentin

DavidGarriou, 2016-12-12:
I understand you problem.
You have this problem because you used an ISR2. In OSEK, an ISR2 is a kind of task, which will be scheduled right after the end of the calling interrupt.
So, in that context, you’ll be out of your interrupt and unable to choose the continuation, i.e. RX or TX treatments.
That is why you need an ISR1, which is an interrupt out of the kernel behaviour. In such an interrupt you have to do very light things.

For the moment continue to use the ISR2 until you driver works properly (as Jean-Luc said before). You’ll be able to modify the kind of ISR after.

If you don’t clear the flag then your interrupt will never be called again.

David

besnarval, 2016-12-12:
I have just done some tests with resources to protect critical sections. I declare a RESOURCE with RESOURCEPROPERTY = STANDARD and add it in my ISR and my task in my .oil file. I called GetResource before my critical section and ReleaseResource after my critical section. Nevertheless, that doesn't work.

With ISR2 and resource: the acknowledgement function for IRQ is called periodically but the ISR is never called (my breakpoint is never reached)
I obtain this backtrace :
#0  HardFault_Handler ()
    at /home/valentin/I3/PFE/trampoline/machines/cortex/armv7em/stm32f407/handlers_stm32f4xx.c:80
#1  <signal handler called>
Backtrace stopped: Cannot access memory at address 0x20020018

With ISR1 and resource: The STM32 crash directly after I start it. I obtain this backtrace :
#0  HardFault_Handler ()
    at /home/valentin/I3/PFE/trampoline/machines/cortex/armv7em/stm32f407/handlers_stm32f4xx.c:80
#1  <signal handler called>
#2  GetResource_exception_call () at uart/tpl_invoque.S:610
#3  0x080029e6 in UART_IRQHandlerTX (uart_id=UART_ID_2) at pfe_uart.c:269
#4  0x080024ee in USART2_IRQ_ClearFlag () at uart.c:25
#5  0x08001870 in USART2_IRQ_Handler () at uart/tpl_app_config.c:578
#6  <signal handler called>
Backtrace stopped: Cannot access memory at address 0x20020018

With ISR2 and resource and treatment for UART done in the ISR (ClearFlag function is empty)
I obtain this backtrace :
#0  HardFault_Handler ()
    at /home/valentin/I3/PFE/trampoline/machines/cortex/armv7em/stm32f407/handlers_stm32f4xx.c:80
#1  <signal handler called>
#2  tpl_save_context ()
    at /home/valentin/I3/PFE/trampoline/machines/cortex/tpl_ctx_switch.S:85
#3  0x08001ae6 in tpl_primary_irq_handler_USART2_IRQ ()
    at uart/tpl_primary_irq.S:407
#4  <signal handler called>
Backtrace stopped: Cannot access memory at address 0x20020018

Best regards

Valentin

jlbirccyn, 2016-12-12:
Could you provide your code ?

besnarval, 2016-12-12:
Yes, I forgot to join the code, sorry.

Here, the code has an ISR2 with critical sections protected by resources and treatment of uart done in the acknowledgement function for IRQ.
[uart_with_resource.zip](https://github.com/TrampolineRTOS/trampoline/files/647537/uart_with_resource.zip)

I just re-run the code on the STM32 but this time, there is no crash. Sometimes, a HardFault was triggered and sometimes not. The result, I obtain is that green led of my task blinks and orange led of my ClearFlag function too. The blue led of the ISR function doesn't blink. I check with GDB and the breakpoint in the ISR function was never reached. 
On my serial interface, I see "abcdefgh" at the beginning and then "**e**bcdefgh" and then "**ef**cdefgh". I checked that with GDB this week end. My "index_write" was not incremented sometimes such that I miss a character. So I think my ISR interrupts the task during execution of a critical section.

PS : With ISR2 and resource and treatment for UART done in the ISR (ClearFlag function is empty), the STM32 crash directly after the first call of the ClearFlag function but I think it is another problem. I have the last backtrace given in my last post.

Valentin

jlbirccyn, 2016-12-13:
Ok

You call GetResource and ReleaseResource from the UART IRQ handler. You cannot do that. Services cannot be called from IRQ Handler except those to disable/enable interrupts. (Get/Release)Resource should be called from within the task or within the ISR. However as stated by David, disable/enable services are not available yet.

So I suggest the following :

* In the USART2_IRQ_ClearFlag, acknowledge the ISR and set a variable to remember if it was a TX or RX IRQ.
* In the ISR, check the variable and process the ISR accordingly. Protect the buffers changes with the resource
* In the TASK, protect the ```UART_puts``` with the source

besnarval, 2016-12-14:
I make the change you advised but that doesn't solve the problem. Could you confirm that I make correctly the change you noticed in your last post ?
Here my code : 
[uart_with_resources_2.zip](https://github.com/TrampolineRTOS/trampoline/files/651812/uart_with_resources_2.zip)

The board crash during the StartOS() function with a UsageFault(). I obtained this backtrace : 
#0  UsageFault_Handler ()
    at /home/valentin/I3/PFE/trampoline/machines/cortex/armv7em/stm32f407/handlers_stm32f4xx.c:101
#1  0x08001d1e in tpl_sc_handler ()
    at /home/valentin/I3/PFE/trampoline/machines/cortex/tpl_system_call.S:232
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Valentin

jlbirccyn, 2016-12-15:
Using the archive you attached to your previous message, I compiled and downloaded the binary to the board and I do not have a crash in ```StartOS```.

I put a breakpoint in StartOS and traced from there : 
* Task Init is scheduled and executes then terminates
* Task idle is scheduled and executes

I continued then crashed from tpl_sc_handler in UsageFault_Handler
At that point the ISR is running (you can check it by printing tel_kern) and by printing the stack frame of the psp, one can see the tel_sc_handler is called from CallTerminateISR2. The crash happens when doing the ```blx r3``` at line 232 of tpl_system_call.S. r3 contains 0 (!?). I tried to put a breakpoint in CallTerminateISR2 to trace the execution but was not able to reproduce the problem. Even after deleting the breakpoint and continuing the execution, it did not crash.

We investigate the problem

jlbirccyn, 2016-12-31:
Hello,

I rewrote many things for the Cortex port. This has been done in the `improved-cortex` branch. In addition, services dealing with interrupts have been implemented.

I suggest you switch to the `improved-cortex` branch by doing `git checkout improved-cortex` to continue your work.

Best regards.

besnarval, 2016-12-31:
Hello,

Thank you very much. I will move on this branch.

Regards

Valentin

besnarval, 2017-01-06:
Hello,

We have tested the UART on the "improved-cortex" branch and it's works very well. We can transmit caracters at 115 200 bauds. 

We have also tested external interrupts. Thats works for EXTI0_IRQ, EXTI1_IRQ, EXTI2_IRQ, EXTI3_IRQ, EXTI4_IRQ. For others, we have a problem of compilation maybe a problem of template in tpl_external_interrupts_c.goilTemplate.

We have also tested timers. The timer 10 works well but when we uses timers 11, 12, 13, 14, we have a problem of compilation. In tpl_app_interrupts_c.goilTemplate, we have a runtime_error when compiling.

We have also tested the SPI and that works well with this branch. Thanks !!

Best regards

jlbirccyn, 2017-01-06:
Hello,

Happy to see we made progress on this issue.

Could you provide projects for configurations that fails ?

besnarval, 2017-01-06:
Hi,

I join these two test projects made by one of my classmate:

- the project with timers
[exemple_timers.zip](https://github.com/TrampolineRTOS/trampoline/files/690011/exemple_timers.zip)

-the project with external interrupts
[extit.zip](https://github.com/TrampolineRTOS/trampoline/files/690012/extit.zip)

Regards

jlbirccyn, 2017-01-08:
Hello,

I just pushed changes related to this issue. Both projects compile now. I added support for UART interrupt dispatch for ISR1 only. In the OIL file, the UART/USART IRQ take a set of sub attributes which are the flags that triggered the IRQ. For instance in USART1, the flags are : TXE, CTS, TC, RXNE, ORE, IDLE, PE, LBD, NF, FE (look at pages 1010-1019 of the reference manual. So if you want to plug an ISR1 on the CTS interrupt of the USART1, you would write in the OIL file:

```
  ISR handle_cts {
    SOURCE = USART1_IRQ { EVFLAG = CTS; };
    CATEGORY = 1;
    PRIORITY = 2;
  };  
```

And if you want to put a second ISR1 on flag LBD, you would write:

```
  ISR handle_lbd {
    SOURCE = USART1_IRQ { EVFLAG = LBD; };
    CATEGORY = 1;
    PRIORITY = 200;
  };
```

You may put more than one flag as sub attribute. In this case the ISR will be call if one or both flags are set.

As a result, goil generate the following IRQ handler:

```
FUNC(void, OS_CODE) USART1_IRQ_Handler(void)
{
  /* ISR1s */
  if ((USART3->SR & (1U << 9)) &&
      (USART3->CR3 & (1U << 10)))
    handle_cts_function();
  if ((USART3->SR & (1U << 8)) &&
      (USART3->CR2 & (1U << 6)))
    handle_lbd_function();
  /* ISR2s */
  /* COUNTERs */
  /* ACK */

  /* Clear flag(s) CTS */
  USART3->SR = ~(1U << 9) & 0x3FFU;
  /* Clear flag(s) LBD */
  USART3->SR = ~(1U << 8) & 0x3FFU;
}
```

```handle_cts_function``` is called only if the flag is set and the interrupts are enabled for this flag. Same for ```handle_lbd_function```. Acknowledge at the end is generated only for flags for which it makes sense to acknowledge them simply. So it is only generated for CTS and LBD. Other flags are cleared by reading/writing other registers or depending on the application behavior. Check page 1012.

*** Not tested, just compiled :) ***

besnarval, 2017-01-10:
Hello,

All timers are now working and all external interrupts are also working. Thanks !
For the UART, there is a problem. It is probably a mistake in the code generated. There is "()" with nothing inside. Here a test project : 
[exemple_uart.zip](https://github.com/TrampolineRTOS/trampoline/files/695669/exemple_uart.zip)

Error with TXE and RXNE flags :
        exemple_uart/tpl_app_config.c: In function 'USART1_IRQ_Handler':
        exemple_uart/tpl_app_config.c:682:18: error: expected expression before ')' token
        USART1->SR = ~() & 0x3FFU;

Regards

Valentin

jlbirccyn, 2017-01-10:
Yes, this line should not be generated because TXE or RXNE are not acknowledged by clearing a bit in SR. I look at that but I will postponed the fix until tonight because goil version has changed and this will disturb the lab if someone does a pull :)

jlbirccyn, 2017-01-10:
Ok, it is fixed. I attach the modified template. You can put it in ```templates/code/cortex/armv7em/stm32f407``` until the changes are pushed

[uart_ack.goilTemplate.zip](https://github.com/TrampolineRTOS/trampoline/files/695751/uart_ack.goilTemplate.zip)

Best regards

besnarval, 2017-01-10:
I have just tested with one ISR on TXE and one ISR on RXNE flag of UART. That works with the template you provide.

Thanks

jlbirccyn, 2017-01-10:
Nice. Just pushed the latest version with reworked templates for interrupt dispatch/acknowledge. Will add ISR2 and COUNTER soon.

closed 2019-10-20 by jlbirccyn