TrampolineRTOS/trampoline #48: The templates path

DC9113, 2016-12-13:
I don't know how to use --templates option or set the GOIL_TEMPLATES environment variable. Where shoud i set this variable?

Snakeuh, 2016-12-13:
Hi !

For an example on how to set the --templates option, you can look at the posix's periodic example : 

> 
> $ cd examples/posix/periodic
> $ goil --target=posix  --templates=../../../goil/templates/ periodic.oil

To set the GOIL_TEMPLATES environment variable, use the "export" command available in any unix shell : 

> $ export GOIL_TEMPLATES="path/to/goil/templates/directory"

If you're lazy like me, from the trampoline directoy : 

> $ cd goil/templates
> $ export GOIL_TEMPLATES="$(pwd -P)"     # "pwd -P" gives you the full physical path of the current directory 

You can put this command in your $HOME/.bashrc file (if you're using a bash shell) so you won't have to set the templates directory anymore : 

> $ cd goil/templates
> $ echo 'export GOIL_TEMPLATES="$(pwd -P)"' >> $HOME/.bashrc

Try the posix's periodic example and let us know if there is any issue.

Kamel Hacene

DC9113, 2016-12-14:
Hi !
Thank you very much for giving me the answer ,but there is still a problem:
I use the goil-windows to compile the periodic.oil file. The results are shown in Figure 1.
Then I move the periodic.oil to "C:\Users\chang\Desktop\TrampolineTry\trampoline-master\goil\templates". The results are shown in Figure 2.

I am a beginner and want to have more communion with you !
[image]
[image]

DC9113, 2016-12-14:
Hi !
Thank you very much for giving me the answer ,but there is still a problem:
I use the goil-windows to compile the periodic.oil file. The results are shown in Figure 1.
Then I move the periodic.oil to "C:\Users\chang\Desktop\TrampolineTry\trampoline-master\goil\templates". The results are shown in Figure 2.

I am a beginner and want to have more communion with you !

[image]

DC9113, 2016-12-14:
Hi, Kamel!
Thank you very much for giving me the answer ,but there is still a problem:
I use the goil-windows to compile the periodic.oil file. The results are shown in Figure 1.
Then I move the periodic.oil to "C:\Users\chang\Desktop\TrampolineTry\trampoline-master\goil\templates". The results are shown in Figure 2.

I am a beginner and want to have more communion with you !
[image]
[image]

jlbirccyn, 2016-12-14:
Hello

According to what I see, it seems you edited periodic.oil and uncommented line 2. Is it ?

DC9113, 2016-12-15:
Hello jlbirccyn: 
Yes, you are right! I did it because results are the same when I use the goil-windows to compile the original file.
What should I do? I have read the Trampoline Handbook carefully but could not find a solution.

jlbirccyn, 2016-12-15:
So comment this line. It is a comment to show the right command to use. It is not intended to be uncommented. 

Since goil.exe is a command-line utility, the easiest way is to make a .bat file to put the command that is fine.

A compileOIL.bat file with this should do the trick:
```
goil.exe --target=posix --templates=../../../goil/templates periodic.oil
pause
```

Attention, to put well the paths that go well (for the folder of the templates in particular). Goil makes the '/' change from the Unix world to the '\' world of Windows.
The 'pause' at the end avoids closing the window.
goil.exe should be in your executable path.

jlbirccyn, 2017-01-09:
Is it ok ? May I close the issue ?

DC9113, 2017-01-10:
Forgive me for not to reply in time.
Actually，the problem has not been solved yet. Would you mind to post more information with attached pictures?
Thank you so much

jlbirccyn, 2017-01-10:
I am lacking informations to help you more. Does goil find its templates or do you have problems later in the build chain ?

Anyway, you cannot compile and execute the POSIX version of Trampoline on Windows because Windows is not POSIX compliant. Some compatibility layers like MinGW will give you most of the POSIX API except the API related to signals which is mandatory.

You can only use it for embedded target on Windows.

If you want to use it on you Windows box, you have to install a Linux in double boot or in a virtual machine using Virtual Box.

Best regards

DC9113, 2017-01-11:
Goil does not find its templates.
Actually, I want to port Trampoline to Freescale MC9S12X128. I plan to use Goil with .oil file to generate files in picture. Then put them together in Codewarrior with other basic files.
The picture shows an example in the path of "trampoline-master\examples\cortex-a\armv7\bcm2836\rpi2\lonely\lonely". How to get them?
[image]

jlbirccyn, 2017-01-11:
Ok,

Where is the trampoline directory on your hard drive ?  from where do you run goil ?

DC9113, 2017-01-12:
Thanks for your question! 
I am too stupid to copy the "goil.exe --target=posix --templates=../../../goil/templates periodic.oil" with no change in compileOIL.bat.
I change the target name like /arch/chip/board and the name of oil file, then I put the compileOIL.bat, goil.exe and XX.oil in the following path "../../../trampolinemaster/goil". Finally several files are generated.
Thank you so much!

closed 2017-01-12 by DC9113