TrampolineRTOS/trampoline #50: Goil error for Raspberry Pi 2

nicolasDSS, 2017-01-15:
Hello, I follow a course you give each week in my school. We use stm32 components and I wanted to try the RTOS on a Raspberry Pi 2 card. I discussed with Mr. Briday and He gave me the path to the rpi examples. I have the following errors when I run this command: 

**********************************************************************************************************
goil --target=cortex-a/armv7/bcm2836/rpi2 --templates=../../../../../../goil/templates/ blink.oil
Created 'blink/tpl_dispatch_table.c'.
Created 'blink/tpl_invoque.S'.
Created 'blink/tpl_os.h'.
Created 'blink/tpl_service_ids.h'.
Created 'build.py'.
Created 'make.py'.
Created 'blink/tpl_app_custom_types.h'.
/home/nicolas/I3/temps_reel/trampoline/goil/templates/code/cortex-a/armv7/bcm2836/counter_call.goilTemplate:23:40:44:
semantic error #1: a string is not a struct
      %  tpl_counter_tick(&% !cpt_fct::VALUE %_counter_desc);
---------------------------------------^^^^^

/home/nicolas/I3/temps_reel/trampoline/goil/templates/code/cortex-a/armv7/bcm2836/counter_call.goilTemplate:23:30:30:
semantic error #2: runtime error
      %  tpl_counter_tick(&% !cpt_fct::VALUE %_counter_desc);
-----------------------------^

/home/nicolas/I3/temps_reel/trampoline/goil/templates/code/cortex-a/armv7/bcm2836/counter_call.goilTemplate:16:3:9:
semantic error #3: runtime error
  foreach cpt_fct in counterFct
--^^^^^^^

/home/nicolas/I3/temps_reel/trampoline/goil/templates/code/cortex-a/armv7/bcm2836/counter_call.goilTemplate:9:1:7:
semantic error #4: runtime error
foreach interrupt in INTERRUPT_SOURCE do
^^^^^^^

/home/nicolas/I3/temps_reel/trampoline/goil/templates/code/tpl_app_config_c.goilTemplate:453:1:8:
semantic error #5: runtime error
template counter_call
^^^^^^^^

/home/nicolas/I3/temps_reel/trampoline/goil/templates/root.goilTemplate:1436:3:10:
semantic error #6: runtime error
  template tpl_app_config_c in code
--^^^^^^^^

/home/nicolas/I3/temps_reel/trampoline/goil/templates/root.goilTemplate:1435:1:5:
semantic error #7: runtime error
write to PROJECT+"/tpl_app_config.c":
^^^^^

Created '/home/nicolas/I3/temps_reel/trampoline/examples/cortex-a/armv7/bcm2836/rpi2/blink/build/blink.oil.dep'.
No warning, 7 errors.

*****************************************************************************************************

May you help me? If I adapt the goil command to another directory like "cortex/armv7m/SmartFusion2/starterKit/blink/" I don't meet any error.

Sincerely,

nicolasDSS.

DavidGarriou, 2017-01-15:
nicolasDSS,
I'll take a look to this problem tonight.
David

DavidGarriou, 2017-01-16:
I have fixed the problem, but i could not test on a rpi2.
This port is not finished, be careful.
I have to work on :
- priorities of interrupts, counters, ISRs
- generation of all external interrupts,

nicolasDSS, 2017-01-17:
Thank you,

I had an error in goil/templates/root.goilTemplates, I removed the last line that was in error (template test in code). So the goil command works but the ./make.py throws an error in a interrupt service. I'll  begin my application without interruption, at first, and I'll add them later.

Thank you for the fixing.

Nicolas.

jlbirccyn, 2017-01-17:
My fault about (template test in code). However I pushed on the improved-cortex branch, not on master. On which branch are you ?

nicolasDSS, 2017-01-17:
I work on improved-cortex.

jlbirccyn, 2017-01-17:
ok. I just merged master into improved-cortex and pushed it. David did the changes in master, not improved-cortex. Do a pull.

closed 2017-01-30 by jlbirccyn