TrampolineRTOS/trampoline #55: Porting Trampoline over PULPino processor

suehtamacv, 2017-05-30:
Hello,

As part of my internship in TIMA laboratory, I'm currently working on porting Trampoline over [PULPino](http://www.pulp-platform.org/), which is based on RISC-V architecture. As my port is still in development, I am following the provided tests to guide my activities.

It has been a while that I'm having problems with the software interruptions, and how to integrate their relative priorities into PULPino's interruption interface. I start to wonder that I shouldn't be using the core's interruption lines to trigger the ISR2s (as by doing so, I might bypass a task of higher priority, as in test hook_s4_full).

I'm interesting into knowing how such interruptions were triggered in other targets else than POSIX. By being close to a task, should an ISR2 only trigger a rescheduling ?

Finally, I would like to ask this issue to remain open, at least until the port is done.

Best regards.

Snakeuh, 2017-05-30:
Hello !

ISRs2 have higher priorities than Tasks, don't worry about bypassing them !

According to PULPino's datasheet, the interrupt priority is managed by their Ids : 

> If multiple interrupts arrive at the same cycle, the interrupt with the lowest number will be executed first. As soon as IRQs are reenabled, either after an eret or after an explicit enable, the interrupt with lowest number will be executed. This means that it is important to clear the interrupt line before reenabling interrupts, as otherwise the same interrupt handler could be called over and over again.

* So you have to generate your tpl_it_table following the interrupt priorities (for instance : ID 0 = Tick interrupts ; ID 1 = Watchdog interrupts ; ID 2 = ISR2 interrupt (priority 6) ; ID 3 = ISR2 interrupt (priority 5), ...). Example : [tpl_init_isr_prio](https://github.com/TrampolineRTOS/trampoline/blob/master/goil/templates/code/ppc/mpc5643l/tpl_external_interrupts_c.goilTemplate) function to see how to handle OIL's priorities ; [tpl_it_table](https://github.com/TrampolineRTOS/trampoline/blob/master/goil/templates/code/ppc/mpc5643l/interrupt_table.goilTemplate) 's generation code.

* Now in your interrupt handler, retrieve the ID of the raised interrupt and jump to the function indexed by this ID in your tpl_it_table. Example : [interrupt handler lines 160 to 203](https://github.com/TrampolineRTOS/trampoline/blob/master/machines/ppc/multicore/tpl_it_handler.s)

* All you have to do in your ISRS2 Handler, is to ack the interrupt and call the kernel's function tpl_central_interrupt_handler_2 with the triggered isrs2's id as a parameter. This call will add an instance of the ISR to the running tasks/isrs2 (just consider the  ISRS2 as being normal Tasks with higher priorities and other privileges for service calls). Example : [tpl_isr_handler](https://github.com/TrampolineRTOS/trampoline/blob/master/goil/templates/code/ppc/mpc5643l/tpl_external_interrupts_c.goilTemplate) functions generation.

* Before returning from the interrupt, check if you need to proceed with a context switch and do it as you would normally with a task's context switch. By doing so, when you'll return from the interrupt, you will execute the isrs2's user function. Example : [interrupt handler lines 250 to 312](https://github.com/TrampolineRTOS/trampoline/blob/master/machines/ppc/multicore/tpl_it_handler.s)

I suggest you to run the code generation of the powerpc multicore examples and take a look at the generated files instead of looking directly at the templates code.

Hope it clears things up.

See you !

suehtamacv, 2017-05-31:
Snakeuh,

Thank you ! By dynamically creating my interrupt table, I managed to solve some of the problems I was having.

There is only a handful of tests that are blocking me (hooks_s4, hooks_s5 and interrupts_s4). I believe that they do not pass because I'm acknowledging the interruptions too soon. I followed [interrupt_table.goilTemplate](https://github.com/TrampolineRTOS/trampoline/blob/master/goil/templates/code/cortex/armv7em/interrupt_table.goilTemplate) from cortex target to create my acking code, and it seems that interruptions are acked whenever their handler is called.

As I saw in one of the examples you sent me, you utilise a return code in your handler to indicate whether we should acknowledge the interruption, avoiding an ISR2 to be acknowledged before they finish their execution. Are those return codes the way to go? Why weren't they needed in Cortex ?

Regards.

Snakeuh, 2017-05-31:
Hi !

Keep in mind that you are in an architecture dependant code. If these implementations differs from one platform to another, it's normal : each architecture manages interrupts its own way.

Now to explain a bit of the powerpc implementation, you need to know that powerpc's interrupts priorities are managed this way : When an interrupt occurs, the cpu will save the interrupt's priority in one of its registers (actually, the interrupt controller does that, but let's make it simple). So when an interrupt with the priority 4 occurs, the processor's priority will switch to the value 4 and disables further interrupts. Now if we enable interrupts without restoring the CPU's interrupt priority, all the interrupts with a priority of 4 and lower will be ignored.

* ISR1 case : The isrs1's user function is called directly in the isrs1's handler. As we are still in kernel mode, Interrupts are disabled during the user function's process. After returning from the handler, we did all we wanted so we restore the cpu's priority.

* ISRS2's case : The isrs2's user function is called after the scheduling, after returning from the interrupt handler. Interrupts are enabled during the process of the user's function, however we don't want interrupts with a priority level inferior than the currently executed isr to slow us. So in fact, we're only restoring the priority level of the cpu later, after the call of the isrs2's user function.

In your architecture, i can see two options : 
* If you can enable interrupts with higher priorities without acknowledging the currently processed isrs2's interrupt : enable higher priorities interrupts before jumping to the isrs2's user function and ack your interrupt after the isrs2's user function call (as it is done in the powerpc's implementation).
* I saw in the user's manual that you could have an interrupt controller that masks interrupts before throwing them to your CPU : after executing the handler of the isrs2's interrupt and before calling the isrs2's user function (before the return from interrupt state so), save the interrupt controller's mask in a software FIFO and manually mask all interrupts with a lower or equal priority. Your FIFO should contain 32 words since you have 32 priority level. After the return of the isrs2's user function, pop and restore the mask.

See you !

suehtamacv, 2017-06-05:
Snakeuh,

Hello !
I thank profusely by your guide. The function `tpl_restore_cpu_priority()` was not implemented, as I did not knew it was called at the end of an ISR2. It made sense that this function was implicated on my problems. My porting seems to be almost complete.

Sadly, I keep having the same problem, with the same test (hook_s4). In this test two software interruptions are trigged, but somehow one of the user tasks is not called (the 6th step does not run). I compared my it_handler to ppc's handler and they are alike. 

Strangely, when I forgot to save context in my it_handler, this specific test went further. It is like I could not depend on `need_save` flag in `tpl_kern` to decide whether to save context or not. But this flag worked well to all the other tests ! This error seems to indicate that my error is in the handler, maybe in the scheduler, but I can't see any further. Maybe my bug description gives you an idea of what's happening ?

Finally, are you interested into adding my port to the ones officially provided ? In this case, are there any guidelines (documentation, code style) that I should follow ? 

Regards.

suehtamacv, 2017-06-09:
Hello,

In file [tpl_os_kernel.c](https://github.com/TrampolineRTOS/trampoline/blob/master/os/tpl_os_kernel.c), line 668, I believe it should be written `s_elected` instead of `s_running`. This seems to be replicated in all branches of Trampoline.

Regards.

jlbirccyn, 2017-06-09:
Hello,

You are probably right. I'll check asap

Best regards

Snakeuh, 2017-06-12:
Sorry for the late reply, i'm a bit short in time for now ! Which test actually fails : hook_s4_non or hook_s4_full ?

I'd say the error may be related to one of these points : 
* You did a context switch after reentering in the kernel : If you enter the kernel twice (i.e a service call made by an ISR), you must wait two exits of the kernel before doing your context switch. Do you have a reentrancy counter ? If not, you may look at the file tpl_it_handler.s in ppc/multicore's implementation, lines 233 for an implementation example.
* Where/when you're disabling/enabling your interrupts in your IT handler.

jlbirccyn, 2017-06-15:
Of course we are interested in adding your port :-)

suehtamacv, 2017-06-15:
Hello,

**It works !**

Thanks for the insights. My problem was indeed in the moment where I reenabled the interruptions. My `tpl_enable_interruptions()` function programmed their reactivation to the *next* return from kernel mode, which worked well unless when this function was called with a reentrancy counter greater than one. This is the reason why interruptions ended up enabled in alarm callbacks... So simple, yet this bug took me weeks to see.

All the tests are passing now. 

Tomorrow I'm going to work in documentation. My work is hosted at https://github.com/suehtamacv/trampoline/tree/pulpino. Should I do a pull request ?

Lastly, feel free to close this issue.

Regards

jlbirccyn, 2017-06-16:
Great !

You can do a pull request when ready to do so. Many thanks for your contribution.

(and Kamel, thanks a lot for the support ;-))

KamelHacene, 2017-06-20:
Good job !

We'll continue this conversation (if needed) in the pull request.
I'll review your contribution this week-end.

See you !

closed 2017-06-20 by KamelHacene