TrampolineRTOS/trampoline #56: [Galgas] Goil compilation fails with a fallthrough (GCC 7.1.1)

Snakeuh, 2017-06-12:
Error when trying to compile goil with gcc (GCC) 7.1.1 20170528 : 

> 
> ../build/libpm/gmp/mpn-tdiv_qr.c: In function ‘__gmpn_tdiv_qr’:
> ../build/libpm/gmp/mpn-tdiv_qr.c:62:7: error: ‘fallthrough’ attribute not followed by ‘;’ [-Werror=attributes]
>        __attribute__ ((fallthrough)) DIVIDE_BY_ZERO;
>        ^~~~~~~~~~~~~
> In file included from ../build/libpm/gmp/mpn-tdiv_qr.c:43:0:
> ../build/libpm/gmp/gmp-impl.h:3934:27: error: this statement may fall through [-Werror=implicit-fallthrough=]
>  #define DIVIDE_BY_ZERO    __gmp_divide_by_zero ()
>                            ^~~~~~~~~~~~~~~~~~~~~~~
> ../build/libpm/gmp/mpn-tdiv_qr.c:62:37: note: in expansion of macro ‘DIVIDE_BY_ZERO’
>        __attribute__ ((fallthrough)) DIVIDE_BY_ZERO;
>                                      ^~~~~~~~~~~~~~
> ../build/libpm/gmp/mpn-tdiv_qr.c:64:5: note: here
>      case 1:
>      ^~~~
>

# Suggestion : 
* Add an attribute fallthrough to the faulty case statement : 
```
diff --git a/goil/build/libpm/gmp/mpn-tdiv_qr.c b/goil/build/libpm/gmp/mpn-tdiv_qr.c
index a2ce07a5..a438c8a7 100644
--- a/goil/build/libpm/gmp/mpn-tdiv_qr.c
+++ b/goil/build/libpm/gmp/mpn-tdiv_qr.c
@@ -60,7 +60,7 @@ mpn_tdiv_qr (mp_ptr qp, mp_ptr rp, mp_size_t qxn,
     {
     case 0:
       DIVIDE_BY_ZERO;
-
+      __attribute__ ((fallthrough));
     case 1:
       {
        rp[0] = mpn_divrem_1 (qp, (mp_size_t) 0, np, nn, dp[0]);
```
* Compile with  "-Wimplicit-fallthrough=0"

jlbirccyn, 2017-06-12:
Hello Kamel

This should be fixed by https://github.com/TrampolineRTOS/trampoline/commit/e81d130e86b5f8211bc950b3e8449b722d27ff57

Snakeuh, 2017-06-12:
Yep, thanks !

closed 2017-06-12 by Snakeuh