TrampolineRTOS/trampoline #69: File 'tpl_os_event_kernel.c' does not compile

kvanherpen, 2017-11-16:
Hi,

When compiling a project using the avr-gcc Compiler (in AtmelStudio), I'm getting a compilation error in the file 'tpl_os_event_kernel.c' (line 200), stating that 'tpl_task_events_table' is undeclared:

\os\tpl_os_event_kernel.c(200,1): error: 'tpl_task_events_table' undeclared (first use in this function)
\os\tpl_os_event_kernel.c(200,1): error: (Each undeclared identifier is reported only once
\os\tpl_os_event_kernel.c(200,1): error: for each function it appears in.)
\os\tpl_os_event_kernel.c(199,1): warning: unused variable 'task_events'

If it is of interest, I also attach the compile command:

"C:\WinAVR-20100110\bin\avr-gcc.exe"  -x c -funsigned-char -funsigned-bitfields -I"../../../../../../os" -I"../../../../../../ioc" -I"../../../../../../machines/avr" -I"../../../../../../com" -I"../../../../../../debug" -I"../../blink" -I"C:\Program Files (x86)\Atmel\Studio\7.0\Packs\atmel\ATmega_DFP\1.2.150\include"  -Os -ffunction-sections -fdata-sections -fpack-struct -fshort-enums -Wall  -mmcu=at90can128  -c -gdwarf-2 -std=gnu99 -MD -MP -MF "tpl_os_event_kernel.d" -MT"tpl_os_event_kernel.d" -MT"tpl_os_event_kernel.o"   -o "tpl_os_event_kernel.o" "../../../../../../os/tpl_os_event_kernel.c"

Could you indicate how I can solve this problem?

Best regards

mbriday, 2017-11-20:
Hi,
The 'tpl_task_events_table' is generated by goil… but only is there are events in your application. If you do not have events, you may remove the file tpl_os_event_kernel.c from your project.

There is a new build system based on python that build the project. This build script is generated by goil and includes only files that should be insreted in your project.

regards

mbriday, 2017-11-20:
We have just updated the goil compiler (3.1.9) to correctly handle paths for the toolchain (note that there are simple quotes "'" for raw strings). 
It has just been successfully tested on Windows with the Atmel Studio embedded AVR toolchain:

```
  OS config {
    STATUS = STANDARD;
    ARDUINO = TRUE {
      SERIAL = TRUE;	//use the serial interface 
    };
    BUILD = TRUE {
      TRAMPOLINE_BASE_PATH = "../../../..";
      APP_NAME = "trampuinoBlink";
      APP_SRC = "blink.cpp";
      CPPCOMPILER = 'C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-g++';
      COMPILER = 'C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-gcc';
      LINKER = 'C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-gcc';
      ASSEMBLER = 'C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-gcc';
      COPIER = 'C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-objcopy';
      SYSTEM = PYTHON;
    };
    SYSTEM_CALL = TRUE;
  };
```
You may now use the python based build scripts (make.py and build.py are generated from goil).
It supports both Python2.7 and 3.x.

The binary file is generated with:
```sh
python make.py
```
regards,

closed 2017-11-20 by mbriday