TrampolineRTOS/trampoline #73: question - Trampoline running on NXPS32k??

emichel11, 2018-01-22:
Hi all,

does anybody know if trampoline is running on NXP S32k?
The processor contains a ARM™ Cortex-M4F/M0+ core, 32-bit CPU
and supports ARMv7 Architecture.
ie similar to STMicroelectronics STM32F4xx or NXP / Freescale MK20DX256.
Maybe anyone has already a port and can share some info with me.
Thanks Edgar

DavidGarriou, 2018-01-22:
Hi Edgar,

the port to this board does not exist. But it shall be easy to port.
Do you have any of this board available ?

David

emichel11, 2018-01-23:
Hello David,
yes - I have a S32K144EVB evaluation board. Any hint which configuration I should take as basis?
 Edgar

DavidGarriou, 2018-01-25:
Hi, 
you should start from one of the Cortex-M4 available board (teensy or discovery).
The core and startup code are the same, you'll have to adapt the code related to the board. 
David