TrampolineRTOS/trampoline #76: NXP DEVKIT-MPC57xx Support.

jed-frey, 2018-11-16:
NXP has released 2 new inexpensive development boards.

- [DEVKIT-MPC5748G, $39.00](https://www.nxp.com/products/processors-and-microcontrollers/power-architecture-processors/mpc5xxx-55xx-32-bit-mcus/ultra-reliable-mpc57xx-32-bit-automotive-and-industrial-microcontrollers-mcus/development-board-for-mpc5748g:DEVKIT-MPC5748G)
  The MPC5748G addresses functional safety, security and Industrial/Automotive gateway application needs, it offers up to three e200 Power Arch MCUs, Hardware Security Module, vast of communication interfaces and flexible memory options.
  DEVKIT-MPC5748G is a development tool for MPC5748G MCU and provides easy access to more than 100 GPIOS of the MCU plus access to onboard Ethernet, CAN, LIN, USB, etc. The standard-based form factor compatible with the Arduino™ pin layout, provides a broad range of expansion board options, the board has option to be powered with external power supply. 
- [DEVKIT-MPC5744P, $35.00](https://www.nxp.com/products/processors-and-microcontrollers/power-architecture-processors/mpc5xxx-55xx-32-bit-mcus/ultra-reliable-mpc57xx-32-bit-automotive-and-industrial-microcontrollers-mcus/development-board-for-mpc5744p:DEVKIT-MPC5744P)
  The MPC5744P addresses Automotive/Industrial functional safety and motor control applications. It offers dual e200z4 lockstep cores, motor control, safety, and communication interfaces to facilitate a complete safety/chassis solution.
  DEVKIT-MPC5744P is a development tool for MPC5744P and provides easy access to more than 100 GPIOs plus access to onboard CAN, LIN, FlexRay, etc. The standard form factor compatible with the Arduino pin layout provides a broad range of expansion options. The board can be powered with an external power supply or directly through USB.

Both are much more accessible options for grad students and hobbyists as the XPC56xxMB2 starts at $375.

Having an actually ASIL certified chip to do development would really lower the barrier of entry for anyone that wanted to jump into AUTOSAR.

jed-frey, 2018-12-17:
I would love to contribute, but my backlog of other projects would probably upset my wife.

Are there any university students (anywhere in the world) that would like to do this as a capstone project? I don't have any connections to any universities but if anyone here does I would happily sponsor some students.

chatoune, 2019-05-07:
Hello,

I have just bought two of those cards (DEVKIT-MPC5744P). I am available to port Trampoline on this platform.
I do not where to start. I have the main.pdf documentation where chapters 15 and 16.2 might be useful.

Regards,

Eddy

jlbirccyn, 2019-05-08:
Hello Eddy,

Good news ! The first thing to do is to run something on the board and to analyze the memory mapping and the startup code. Since this CPU supports only the VLE ISA, you will start by doing a copy of machines/ppc/mpc5643l and rename it to mcp5744p (startup code) and goil/templates/code/ppc/mpc5643l (templates to generate target specific code) and goil/templates/config/ppc/mpc5643l (basically the files to compile in machines/ppc/mpc5643l and interrupt sources). What compiler are you going to use ?

chatoune, 2019-05-08:
Hello,

I will be using GCC. I am working with the IDE (S32 DS) from NXP.
I have also added the latest SDK which includes drivers for the MPC5744P and also GCC (4.9.4).

jlbirccyn, 2019-05-09:
Hello,
So, in addition, in goil/templates/linker/gnu_ld/ppc/mpc5744p you will have to provide a link script template. It will be similar to the one in mpc5510 but with a different memory mapping. But I attach a link script template for the MPC5643L and GNU LD below

[script.goilTemplate.zip](https://github.com/TrampolineRTOS/trampoline/files/3160687/script.goilTemplate.zip)

The template language has a documentation you'll find here : https://github.com/TrampolineRTOS/GTL/blob/master/documentation/GTL.pdf

Best regards

chatoune, 2019-05-09:
Hi,

Thanks for your support. I have forked the project and I will update it tonight with all the modifications. I downloaded the complete documentation set for MPC5744P and MPC5643L.
For the link script, NXP is providing examples for this platform. With the GTL doc I might be able to have a first version quickly.

Regards,

Eddy

jed-frey, 2019-06-30:
@chatoune , let me know if you need extra testers.

chatoune, 2019-07-03:
@jed-frey , thanks for your proposal.
I had some work priorities for few weeks, now that I am available again I will rework my code (and write a complete porting documentation). I am still missing the linker script and then we could test it.

Regards,

Eddy

chatoune, 2019-07-29:
Hi, just to give some news about the project. I am reviewing the scripts to verify I am not missing anything.
I noticed boundaries which are not tested.
In the root.goilTemplate, priority limits should be tested. If I choose 2<sup>32</> as a priority then the result is over the limit of an UINT32 and there is no place for the ISRs.

jlbirccyn, 2019-07-29:
Hello,
In fact priorities are packed by the script so that there is no hole in priority levels. So even if you use 2<sup>32</sup> for a task priority and if this task is alone, its priority will be in fact 1 (0 being the priority of the idle task). However you are right, even if 2<sup>32</sup> tasks would be needed to overflow the priority levels and would overflow the RAM of any microcontroller, this should be checked.

mahmudsamiunlu, 2020-01-03:
Hello, did you figure out MPC5744P GCC linker script for autosar mcal 4.0 ? Can you help me about linker script ?

quyleduc, 2022-10-20:
Anyone have update for this topic?It's very interesting