TrampolineRTOS/trampoline #84: AVR new libraries don't compile

taha-ub2, 2020-05-06:
Hi every one 
I am trying to use trampoline with Arduino, but when it comes to add new libraries such as LiquidCrystal_I2C or servo it doesn't compile  
if any one can help please

dapperfu, 2020-05-06:
"It doesn't compile" is a very broad error.

1. What is the error message?
2. What have you tried so far?

mbriday, 2020-05-07:
Hi,
Can you give me more details about the compilation problem? 
regards

taha-ub2, 2020-05-07:
Hi, 
the problem is that I try to make a task that displays data from sensors for the first time I want just to test the LCD so for that I have to add the LiquidCrystal_I2C library, for that I modified the config.oil as mentioned but it tells me I need a Wire.h file when i added Wire.h into the library folder 
here Iput  a  zip file with oil and cpp files and screenshots on the errors I got
[lcd.zip](https://github.com/TrampolineRTOS/trampoline/files/4593596/lcd.zip)

mbriday, 2020-05-07:
Hi,
Ok, received. I will look at that asap.

mbriday, 2020-05-08:
Hi,

I tried your code. In fact, you try to insert a new library, but it depends on i2c (Wire). So you just have to set it in your project:
```      
   LIBRARY = lcd_i2c;
   LIBRARY = i2c;
```

Another way is to define the dependency directly in the config.oil:

```
  LIBRARY lcd_i2c {
    NEEDS   = i2c;
    PATH    = "/avr/arduino/libraries/LiquidCrystal_I2C-1.1.2";
    CPPFILE = "LiquidCrystal_I2C.cpp";
  };
```
This is the preferred approach, as i2c will be added automatically. I have updated the README.md in `example/avr` to add the information about the `NEED` parameter.

There is still a problem, with a conflict with the `write()` function. I look for this pb. Maybe the arduino version in Trampoline should be updated.

regards,

Note: In your example, you just have one task, that is not AUTOSTART and will never be activated.

mbriday, 2020-05-08:
in the library header file, there is the following code:

```c
#if defined(ARDUINO) && ARDUINO >= 100
  virtual size_t write(uint8_t);
#else
  virtual void write(uint8_t);
#endif
```
So, you have to define the ARDUINO definition. You can add C/C++ flags in the `BUILD` section of your .oil file:

```
      APP_SRC     = "lcd.cpp";
      CFLAGS      = "-DARDUINO=10809";
      CPPFLAGS    = "-DARDUINO=10809";
      CPPCOMPILER = "avr-g++";
```
And it just works :) (in fact, it compiles, I have no i2c based lcd here)

Note: I have updated the arduino submodule, you can update.

taha-ub2, 2020-05-08:
Hi Mr mbriday
I have done exactly as you said but still have a problem 
thank you very much for your help

[image]

mbriday, 2020-05-08:
This is a link problem.
Print::print is not found. It is defined in `Print.cpp`. It is added in the `serial` lib.
It just made an update to split serial in `print` and `serial`. Make an update and declare your library as:

```
  LIBRARY lcd_i2c {
    NEEDS   = i2c;
    NEEDS   = print;
    PATH    = "/avr/arduino/libraries/LiquidCrystal_I2C-1.1.2";
    CPPFILE = "LiquidCrystal_I2C.cpp";
  };
```
In my working example, I had serial declared.
regards,

taha-ub2, 2020-05-08:
Hi Mr Mikael 
first I am very grateful it finally worked 

  LIBRARY lcd_i2c {
    NEEDS   = i2c;
    NEEDS   = serial;  // here i did serial instead of print
    PATH    = "/avr/arduino/libraries/LiquidCrystal_I2C-1.1.2";
    CPPFILE = "LiquidCrystal_I2C.cpp";
  };
and in the cpp file I have to include LiquidCrystal_I2C.h
I wish if I could just  add libraries on the build and not modify  the cpp files.
thank you very much 
best regards.

mbriday, 2020-05-09:
I didn't have to modify the library sources, but I don't think it is a Trampoline related problem anymore.
Regards,

closed 2020-05-09 by mbriday