TrampolineRTOS/trampoline #85: driver with task/isr.

mbriday, 2020-05-15:
The goil templates allows to declare a driver through the LIBRARY object (in config.oil file). However, the LIBRARY object embeds a path and a set of functions (header/code). 

it is useful for basic drivers, but it may be improved with the support of an OIL subset (isr, tasks, alarms,…) for more complex drivers.

mbriday, 2020-05-16:
In fact, we just need to declare an extra OIL file, like that:
```
  LIBRARY serial {
    PATH = "cortex/armv7em/stm32f303/lib";
	CHEADER = "serial.h";
	CFILE   = "serial.c";
	OILFILE = "serial.oil";
  };

```

jlbirccyn, 2020-05-16:
Yes but since the goil has no idea of the meaning of the OILFILE attribute, serial.oil will not be included and an OIL file cannot be included by a goil template because when the template code is executed the OIL parsing is finished. So we have to be able to specify the OILFILE corresponds to an include and that means modifying goil.

mbriday, 2020-05-16:
Yes, we need to call the parser again.
Will it be possible to extend gtl so that it can call the parser ?

jlbirccyn, 2020-05-16:
Everything is possible but it is a lot of work. We have to parse an OIL file and do not call GTL, instead return the whole dataset normally passed to root.goilTemplate. Get the result in  map (at least the data type already exist in GTL. GTL has to know about the OIL parser.