TrampolineRTOS/trampoline #87: License question

ldearquer, 2020-10-06:
I have seen that the project license has moved from LGPL to GPL some time ago, so wanted to ask: are software components on the application layer affected by this?
I understand distributing Trampoline as is, or with any changes, requires to release the source code, but I was wondering if it was meant to affect only the OS itself, and not any application (kind of what the Linux kernel explicitly states on its license).
I guess the technical question would be what the copyright holders consider "derivative work" under the GPL terms.
Thanks!
Luis

jlbirccyn, 2020-10-07:
Hello,

Any application, as it is statically linked to Trampoline, must be under a GPL-compatible license. This is the difference between LGPL and GPL, the LGPL having an exception for linking.

However, our employers (CNRS, University of Nantes and Ecole Centrale de Nantes) have delegated the commercial management of Trampoline to "SATT Ouest Valorisation" and an ad-hoc licence is possible within a commercial framework. https://www.ouest-valorisation.fr/en/home-page/

ldearquer, 2020-10-07:
OK, thanks for the quick reply

closed 2021-05-20 by jlbirccyn