TrampolineRTOS/trampoline #88: Problème avec STM32F4 (config windows) -> no rule for making "tpl_os_kernel.c"

KacimiBadr, 2020-10-09:
**Bonjour,
Je me bats depuis quelques jours avec mon ordinateur pour essayer de faire fonctionner trampoline sur la carte STM32f4, j'ai suivi les instructions suivantes  : https://www.docdroid.net/CwljIDq/manuel-dinstallation-stm-windows-pdf

goil.exe ne produit aucune erreur, mais l'instruction "build python.py" produit le message d'erreur suivant :** 

F:\RessourcesLabsSTM32\trampoline-master\trampoline-master\labs\labs_stm32F4_discovery\lab1>python build.py
[1m[91mNo rule for making 'F:RessourcesLabsSTM32/trampoline-master/trampoline-master/os/tpl_os_kernel.c'[0m 
**(j'ai fait un copier coller de la fenêtre de la console cmd )**

**Avant cela j'avais eu un souci avec Python et un message d'erreur concernant l'absence du module makefile et projfile, mais j'avais réussi à régler le problème je ne sais pas comment précisément.

L'erreur "No rule for making ... tpl_os_kernel.c" comment de résoudrait-elle ?
Merci d'avance.
Cordialement,**

jlbirccyn, 2020-10-09:
Bonjour

C'est ```python make.py``` qu'il faut lancer.

KacimiBadr, 2020-10-09:
Merci pour la réponse,
Mais voici le résultat : 

```
F:\RessourcesLabsSTM32\trampoline-master\trampoline-master\labs\labs_stm32F4_discovery\lab1>python make.py
[1m[94mNothing to make.[0m
[1m[91mNo rule for making 'F:RessourcesLabsSTM32/trampoline-master/trampoline-master/os/tpl_os_kernel.c'[0m
```
(Copier coller de la fenêtre console)

Je peux fournir plus d'informations si cela peut aider à identifier le problème, si besoin.

KacimiBadr, 2020-10-09:
Voici make.py (https://pastebin.com/LUj5ABED) et voici build.py (https://pastebin.com/fUUkAJXY). Cdlt

mbriday, 2020-10-09:
Je suis en train de démarrer une VM Windows pour regarder… mais ça se trouve je peux reproduire sur mon Linux. Je teste de suite.

mbriday, 2020-10-09:
Je viens de tester sous Linux. Pas de problème de compilation. Voici le test réalisé:

Aller dans un dossier d'exemple du STM32F407:
```
cd examples/cortex/armv7em/stm32f407/stm32f4discovery/blink
```
Lancer goil avec les bon paramètres (il sont indiqués dans le README.md):
```
goil --target=cortex/armv7em/stm32f407/stm32f4discovery --templates=../../../../../../goil/templates/ blink.oil
```
Puis compilation avec le script de construction:
```
./make.py
```

Avez vous fait des étapes comme ici? (notamment la définition de la cible `--target=…`

Note: La première fois: s'il ne trouvait pas le module makefile, c'est probablement que le dossier d'installation de trampoline n'était pas correctement renseigné dans le fichier .oil

jlbirccyn, 2020-10-09:
Je soupçonne un problème de chemins Windows dans les scripts Python. Notamment, je vois dans build.py des chemins spécifiés comme ceci : ```F:RessourcesLabsSTM32/trampoline-master/trampoline-master/```. C'est normal en Python sous Windows ?

KacimiBadr, 2020-10-09:
Oui il me semble, c'est l'étape 9 du fichier PDF décrivant les étapes de config windows (lien du premier post). On peut voir dans les lignes suivantes, que tous les chemins sont "inclus" , je vois en ligne  129 l'instruction : 
(includeDirs += ["-I", "F:RessourcesLabsSTM32/trampoline-master/trampoline-master/os"]) le chemin contenant le fichier "tpl_os_kernel.c" mentionné dans le titre.

Pour ce qui est du goil voici l'instruction et son résultats : 
```
F:\RessourcesLabsSTM32\trampoline-master\trampoline-master\labs\labs_stm32F4_discovery\lab1>goil.exe --target=cortex/armv7em/stm32f407/stm32f4discovery --templates=/RessourcesLabsSTM32/trampoline-master/trampoline-master/goil/templates lab1.oil
Replaced 'lab1/tpl_dispatch_table.c'.
Replaced 'lab1/tpl_invoque.S'.
Replaced 'lab1/tpl_os.h'.
Replaced 'lab1/tpl_service_ids.h'.
Replaced 'build.py'.
Replaced 'make.py'.
Replaced 'lab1/tpl_app_custom_types.h'.
Replaced 'lab1/tpl_app_config.c'.
Replaced 'lab1/tpl_app_config.h'.
Replaced 'lab1/tpl_app_define.h'.
Replaced 'lab1/MemMap.h'.
Replaced 'lab1/Compiler.h'.
Replaced 'lab1/Compiler_Cfg.h'.
Replaced 'lab1/script.ld'.
Replaced 'lab1/AsMemMap.h'.
Replaced 'lab1/tpl_static_info.json'.
Replaced 'lab1/tpl_vectors.c'.
Replaced 'lab1/tpl_primary_irq.S'.
Replaced 'lab1/cmsis_wrapper.h'.
Replaced 'lab1/tpl_external_interrupts.c'.
Replaced 'lab1/tpl_app_interrupts.c'.
Replaced 'lab1/tpl_cortex_definitions.h'.
Replaced 'lab1/stm_structure.c'.
Replaced '/F/RessourcesLabsSTM32/trampoline-master/trampoline-master/labs/labs_stm32F4_discovery/lab1/build/lab1.oil.dep'.
No warning, no error.

```

Je vais à présent décrire ma config complètement de A à Z. juste un instant que je puisse l'écrire

mbriday, 2020-10-09:
ou un problème de version de goil?

mbriday, 2020-10-09:
Bon, je suis dans les essais sous Windows. Le fichier .oil n'aime pas du tout les chemins bizarres (comme 'Program files (x86)'). les parenthèses ne sont pas prévues car non utilisées sous Unix

KacimiBadr, 2020-10-09:
Oui, voici la description complète de mon projet.
Premièrement une vue sur le dossier "RessourcesLabsSTM32" [https://i.postimg.cc/mkFWzBpx/screenshot-304.png](https://i.postimg.cc/kMBkBqD1/screenshot-305.png)
1) Pour ce qui est de TRAMPOLINE_BASE_PATH, il s'agit du dossier qui pointe vers les dossiers : autosar, goil, machines.. etc, donc pour l'exemple du PDF d'instructions (dans le premier post) c'est directement "/RessourcesLabsSTM32/trampoline", mais pour moi c'est : 

`TRAMPOLINE_BASE_PATH = "F:RessourcesLabsSTM32/trampoline-master/trampoline-master";`

Vu que j'ai téléchargé Trampoline depuis ce lien : https://github.com/TrampolineRTOS/trampoline/archive/master.zip

voici le contenu de labs1.oil (https://pastebin.com/Psp8CNGU), cela va être plus clair dans la description du projet qui suit.

2) Pour comprendre le dossier labs et toute la structure du projet, je vais simplement partager avec vous le dossier labs sur ce lien : `https://mega.nz/folder/lQFgXDxT#Di9HU84D4cf-Vtbihphvug`
Le dossier labs, je l'ai mis dans TRAMPOLINE_BASE_PATH, c'est à dire ( F:\RessourcesLabsSTM32\trampoline-master\trampoline-master\labs), donc pour effectuer les différentes tâches de ce TP il suffit de se balader sur labs1 labs2 etc  
Par exemple, labs1 : F:\RessourcesLabsSTM32\trampoline-master\trampoline-master\labs\labs_stm32F4_discovery\lab1>
à partir d'ici nous effectuons 3 lignes de commande 
A) goil.exe --target=cortex/armv7em/stm32f407/stm32f4discovery --templates=/RessourcesLabsSTM32/trampoline-master/trampoline-master/goil/templates lab1.oil
B) python build.py
C) python build.py burn (pour transférer sur la carte STM32)

Moi mon message d'erreur a lieu dans l'étape 2 "python build.py"

Voici une image pour montrer l'emplacement de labs : [https://i.postimg.cc/cJFbLbWG/screenshot-306.png](https://i.postimg.cc/cJFbLbWG/screenshot-306.png)
3) Le lien mega ci-dessus, contient également un exécutable goil fournit par les responsables du projet labs que je suis en train de réaliser, il semble que l'utilisation d'une version récente de goil ne marche pas, il faut respectivement : A) télécharger goil sur (Le lien http://trampolinebin.rts-software.org/goil-windows.zip) décompresser et l'ajouter dans un dossier (F:\RessourcesLabsSTM32\goil-windows)  (tout ajoutant le PATH windows variable d'environnement), ensuite B)  SUPPRIMER l'exe goil téléchargé, ne garder que "goil-debug" dans le dossier, ensuite C) récupérer le fichier exe goil fournit par les responsables du projet labs et le coller dans l'emplacement F:\RessourcesLabsSTM32\goil-windows.
De cette manière la ligne de commande (goil.exe --target=cortex/armv7em/stm32f407/stm32f4discovery --templates=/RessourcesLabsSTM32/trampoline-master/trampoline-master/goil/templates lab1.oil) fonctionne.

4)  Deux fichiers (tp_ups.c et tp_ups.h) sont copiés et collés sur  (« F:\RessourcesLabsSTM32\trampoline\machines\cortex\armv7em\stm32f407\stm32f4discovery »), c'est l'étape 8 du PDF manuel d'installation STM windows ( lien dans le premier post).

[https://i.postimg.cc/fbPfFnZy/screenshot-307.png](https://i.postimg.cc/fbPfFnZy/screenshot-307.png)
5) pour ce qui de STLINK, j'ai téléchargé la version mentionné dans le PDF : (https://github.com/texane/stlink/releases/download/1.3.0/stlink-1.3.0-win64.zip)

6) pour ce qui est du compilateur Arm pour processeurs Cortex-M (GNU Arm Embedded Toolchain), j'ai téléchargé la dernière version mais en 32 bits bien sur, celui ci : "gcc-arm-none-eabi-9-2020-q2-update-win32", sauf que j'ai effectué un changement de nom du dossier en : "GNU_Tools_ARM_Embedded"

7) Enfin, pour ce qui est de python, j'ai supprimé toutes les anciennes versions que j'avais et j'ai installé python 2.7, j'ai ajouté les variables PATH windows : [https://i.postimg.cc/kMBkBqD1/screenshot-305.png](https://i.postimg.cc/kMBkBqD1/screenshot-305.png)

Voici labs1.c pour reference (mais vous pouvez le trouver dans le dossier labs que je vous ai partagé) https://pastebin.com/wmnFbHRu

Remarque importante? Puisque je suis bloqué à la seconde ligne de commande "python build.py", il semble les erreurs sur le fichier lab1.c ne sont pas détectés (car il n'est pas compilé?) j'ai essayé d'introduire des erreurs mais elle n'ont pas été vues.

mbriday, 2020-10-09:
Est-ce que vous pouvez essayer:

- avec la dernière version de Trampoline (le master du dépôt Git)
- avec la version de goil ci jointe [goil.exe.zip](https://github.com/TrampolineRTOS/trampoline/files/5356529/goil.exe.zip)
- en mettant tout sur le même lecteur C:\ ou F:\

KacimiBadr, 2020-10-09:
- D'accord.
- Entendu
- C'est à dire, sans passer pas le dossier "RessourcesLabsSTM32" ? Donc au lieu d'avoir (F:\RessourcesLabsSTM32\trampoline-master\trampoline-master\labs\labs_stm32F4_discovery\lab1>) ça sera (F:\trampoline-master\trampoline-master\) ?
Quid des restes dossiers https://i.postimg.cc/mkFWzBpx/screenshot-304.png ?

mbriday, 2020-10-09:
Non, pas de référence à un lecteur 'F:\'. Donc tous les fichiers doivent être sur le même lecteur.

KacimiBadr, 2020-10-09:
YES! ça a marché ! Mille fois merci !
J'ai du réarranger tout mon F pour être sur qu'il soit encore lisible.
Pour résumer, pour les prochains lecteurs :
1) J'ai mis tous les fichiers de trampoline (autosar etc..) dans F:, comme demandé par mbriday.
2) J'ai fait de même pour les fichiers de ARM (arm-none-eabi, bin.. etc)
3) j'ai ajouté labs et python 27.

ça marche! Il a fallu lancer "python build.py burn" 2 fois, et le programme a fonctionné.

Merci encore @mbriday. Et excellent week end!

mbriday, 2020-10-10:
Bonne nouvelle!

Il y a donc un problème avec la gestion de lecteur comme l'indiquait @jlbirccyn!

Quand je disais de mettre les fichiers sur le même lecteur, je sous-entendais les fichiers manipulés par goil et/ou python, donc les fichiers sources (.c/.h/.s)net de configuration (.oil+templates). Par contre, les binaires  (compilateur, stlink et python) peuvent être installés n'importe où (comme on utilise la variable d'environnement pour les trouver)

Dernier point, il n'est pas indispensable d'utiliser une version obsolète de python. Python3 fonctionne sans problème.

Cordialement,

closed 2020-10-10 by mbriday