TrampolineRTOS/trampoline #89: arm/nxt/demo_car complain goil version dismatch

JerryFan0913, 2020-10-17:
I fetch latest code from master branch, and try to build this demo but get the complain like below: 

compiling goil file bluetooth2.oil
goil --target=arm/nxt --templates=/mnt/d/Work/Repo/trampoline/goil/templates  bluetooth2.oil
/mnt/d/Work/Repo/trampoline/goil/templates/root.goilTemplate:33:3:7:
semantic error #1: Goil version 3.1.11 required
  error here : "Goil version " + NEEDED_VERSION + " required"

I checked the history of trampoline/goil/templates, and seems that it requires goil 3.1.11 since 2010 Oct. Unfortunately, the goil package provided in trampoline git page is 3.1.9. Any idea to fix this? Or where I can get the latest goil?

JerryFan0913, 2020-10-17:
Not noticed that the goil source code had been included in trampoline and can be built there. This is not a real issue actually.

closed 2020-10-17 by JerryFan0913

jlbirccyn, 2020-10-17:
Goil build binaries are lagging a little bit. Will try to fix that next week. Glad to see you found the sources. 

Best regards