TrampolineRTOS/trampoline #90: goil throw errors when dealing with examples/cortex-a/armv7/bcm2836/rpi2/blink

JerryFan0913, 2020-10-19:
How to reproduce it: 
#1. On Linux, build goil using latest code.
#2. Under examples/cortex-a/armv7/bcm2836/rpi2/blink， run “goil --target=cortex-a/armv7/bcm2836/rpi2 --templates=../../../../../../goil/templates/ blink.oil”， and below erros thrown by goil: 

jerryfan@DESKTOP-9PNJKVH:/mnt/d/Work/Repo/trampoline/examples/cortex-a/armv7/bcm2836/rpi2/blink$ goil --target=cortex-a/armv7/bcm2836/rpi2 --templates=../../../../../../goil/templates/ blink.oil
Replaced 'blink/tpl_dispatch_table.c'.
Replaced 'blink/tpl_invoque.S'.
Replaced 'blink/tpl_os.h'.
Replaced 'blink/tpl_service_ids.h'.
/mnt/d/Work/Repo/trampoline/goil/templates/build/build_py.goilTemplate:664:37:43:
semantic error #1: Variable or field does not exist
      print("\033[1m\033[95m% !cmd::MESSAGE % % !OS::BUILD_S::APP_NAME % on % ! TARGET %\033[0m")
------------------------------------^^^^^^^

/mnt/d/Work/Repo/trampoline/goil/templates/build/build_py.goilTemplate:664:31:31:
semantic error #2: runtime error
      print("\033[1m\033[95m% !cmd::MESSAGE % % !OS::BUILD_S::APP_NAME % on % ! TARGET %\033[0m")
------------------------------^

/mnt/d/Work/Repo/trampoline/goil/templates/build/build_py.goilTemplate:624:3:9:
semantic error #3: runtime error
  foreach cmd in pc::COMMAND do

mbriday, 2020-10-19:
Hi,
This has just been fixed in commit 6f436b2. Please git pull and try again.
regards

JerryFan0913, 2020-10-22:
many thanks for the hot fix. @mbriday

closed 2020-10-22 by mbriday