TrampolineRTOS/trampoline #92: goil compilation fail

massimoviolante, 2020-10-29:
When compiling goil (under Ubuntu Linux 18.04 x64 and MacOSX), the following errors happen. Any idea? Thanks.

Making "../build/cli-objects/makefile-macosx-objects" directory
[  0%] Native Compiling for OS X: all-galgas.cpp
[  0%] Native Compiling for OS X: all-libpm.cpp
[  1%] Native Compiling for OS X: all-declarations-12.cpp
[  1%] Native Compiling for OS X: all-declarations-11.cpp
[  1%] Native Compiling for OS X: all-declarations-10.cpp
[  2%] Native Compiling for OS X: all-declarations-9.cpp
[  2%] Native Compiling for OS X: all-declarations-8.cpp
[  3%] Native Compiling for OS X: all-declarations-7.cpp
[  3%] Native Compiling for OS X: all-declarations-6.cpp
[  3%] Native Compiling for OS X: all-declarations-5.cpp
[  4%] Native Compiling for OS X: all-declarations-4.cpp
[  4%] Native Compiling for OS X: all-declarations-3.cpp
[  4%] Native Compiling for OS X: all-declarations-2.cpp
[  5%] Native Compiling for OS X: all-declarations-1.cpp
[  5%] Native Compiling for OS X: all-declarations-0.cpp
[  6%] Native Compiling for OS X: all-predefined-types.cpp
[  6%] Native Compiling for OS X: C_galgas_quiet_option.cpp
[  6%] Native Compiling for OS X: 32-mp_bases.c
[  7%] Native Compiling for OS X: mpn-invertappr.c
[  7%] Native Compiling for OS X: mpn-sub_1.c
[  8%] Native Compiling for OS X: mpn-toom_eval_dgr3_pm1.c
[  8%] Native Compiling for OS X: mpz-clrbit.c
[  8%] Native Compiling for OS X: mpz-set_ui.c
[  9%] Native Compiling for OS X: mpn-lshift.c
[  9%] Native Compiling for OS X: mpn-sub_n.c
[  9%] Native Compiling for OS X: mpn-toom_eval_dgr3_pm2.c
[ 10%] Native Compiling for OS X: mpz-cmp.c
[ 10%] Native Compiling for OS X: mpz-setbit.c
[ 11%] Native Compiling for OS X: 64-mp_bases.c
[ 11%] Native Compiling for OS X: mpn-lshiftc.c
[ 11%] Native Compiling for OS X: mpn-submul_1.c
[ 12%] Native Compiling for OS X: mpn-toom_eval_pm1.c
[ 12%] Native Compiling for OS X: mpz-com.c
[ 12%] Native Compiling for OS X: mpz-sizeinbase.c
[ 13%] Native Compiling for OS X: mpn-mu_div_qr.c
[ 13%] Native Compiling for OS X: mpn-tdiv_qr.c
[ 14%] Native Compiling for OS X: mpn-toom_eval_pm2.c
[ 14%] Native Compiling for OS X: mpz-combit.c
In file included from ../build/output/all-predefined-types.cpp:10:
../build/libpm/galgas2/capSortedListElement.h:33:7: error: class member cannot be redeclared [-Werror,-Wredeclared-class-member]
class cSortedListElement ;
      ^
../build/libpm/galgas2/cSortedListElement.h:31:7: note: previous declaration is here
class cSortedListElement : public cCollectionElement {
      ^
In file included from ../build/output/all-declarations-4.cpp:8:
In file included from ../build/output/all-declarations-8.cpp:8:
In file included from ../build/output/all-declarations-0.cpp:8:
In file included from ../build/output/all-declarations-4.h:9:
In file included from ../build/output/all-declarations-8.h:9:
../build/output/all-declarations-0.h:135:1: error: storage class specified for a member declaration
In file included from ../build/output/all-declarations-3.h:9:
In file included from ../build/output/all-declarations-7.h:9:
In file included from ../build/output/all-declarations-2.h:9:
In file included from ../build/output/all-declarations-1.h:9:
In file included from ../build/output/all-declarations-6.h:9:
extern const C_galgas_type_descriptor kTypeDescriptor_GALGAS_gtlDataList ;
^
In file included from ../build/output/all-declarations-2.cpp:8:
In file included from ../build/output/all-declarations-5.h:9:
In file included from ../build/output/all-declarations-1.cpp:8:
In file included from ../build/output/all-declarations-3.cpp:8:
../build/output/all-declarations-0.h:135:1: error: storage class specified for a member declaration
In file included from ../build/output/all-declarations-2.h:9:
In file included from ../build/output/all-declarations-1.h:9:
extern const C_galgas_type_descriptor kTypeDescriptor_GALGAS_gtlDataList ;
^
[... 1255 lines cut ...]
In file included from ../build/output/all-declarations-4.h:9:
In file included from ../build/output/all-declarations-3.h:9:
In file included from ../build/output/all-declarations-2.h:9:
In file included from ../build/output/all-declarations-1.h:9:
../build/output/all-declarations-0.h:699:7: error: 'GALGAS__32_lstringlist_2D_element::cMapElement_gtlExpressionMap' has virtual functions but non-virtual destructor [-Werror,-Wnon-virtual-dtor]
class cMapElement_gtlExpressionMap : public cMapElement {
      ^
../build/output/all-declarations-0.h:787:1: error: storage class specified for a member declaration
extern const C_galgas_type_descriptor kTypeDescriptor_GALGAS_gtlExpressionMap_2D_element ;
^
../build/output/all-declarations-0.h:919:1: error: storage class specified for a member declaration
extern const C_galgas_type_descriptor kTypeDescriptor_GALGAS_gtlInstructionList ;
^
../build/output/all-declarations-0.h:969:1: error: storage class specified for a member declaration
extern const C_galgas_type_descriptor kTypeDescriptor_GALGAS_gtlInstruction ;
^
../build/output/all-declarations-0.h:1060:1: error: storage class specified for a member declaration
extern const C_galgas_type_descriptor kTypeDescriptor_GALGAS_gtlInstructionList_2D_element ;
^
../build/output/all-declarations-0.h:1072:1: error: storage class specified for a member declaration
extern const char * kSearchErrorMessage_gtlTemplateMap_get ;
^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
1 error.

mbriday, 2020-10-29:
Hi,
Can you give us the MAC OS X version, the XCode version and the gcc (llvm) version?
What is the command you entered.
regards,

massimoviolante, 2020-10-29:
$ uname -a
Darwin Kernel Version 19.6.0: Mon Aug 31 22:12:52 PDT 2020; root:xnu-6153.141.2~1/RELEASE_X86_64 x86_64

$ gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 12.0.0 (clang-1200.0.32.21)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

I cloned the master branch few weeks ago and it worked flawlessly. I tried today with an older branch (trace) and everything worked fine. 
Regards

mbriday, 2020-10-29:
Ok. So, If I understand well, you tried with the up-to-date master branch?
I will test as soon as I have a Mac nearby.
regards,

massimoviolante, 2020-10-29:
Thank you. Same problem happens under Ubuntu Linux 18.04 x64.

sefau, 2020-10-29:
Hi.
I have also a compilation problem (albeit different) on Debian 10  with gcc (Debian 10.2.0-15) 10.2.0.
While we investigate, notice that you should still be able to use an older version of goil to build your projects.
Regards.

mbriday, 2020-10-29:
I have just pushed a fix. This now works for me on Ubuntu 18.04. Can you pull, test and let me know?
regards,

massimoviolante, 2020-10-29:
That was fast! I checked the compilation under both MacOSX and Ubuntu and it worked flawlessly. I also checked the periodic example and it worked as well. Thank you very much!

closed 2020-10-29 by massimoviolante

jlbirccyn, 2020-10-29:
Just did the same with Galgas 3.4.1 😉

jlbirccyn, 2020-10-30:
> Hi.
> I have also a compilation problem (albeit different) on Debian 10 with gcc (Debian 10.2.0-15) 10.2.0.
> While we investigate, notice that you should still be able to use an older version of goil to build your projects.
> Regards.

Should be ok with commit cc42546071f11035a99c985c813493e3b34d47d0