TrampolineRTOS/trampoline #93: [Arduino UNO] Using another timer instead of Timer0 for OSEK's Counters

thegabriele97, 2020-11-10:
Hi,
I would like to use another timer for a personalized counter instead of the predefined one (SystemCounter). 
I aimed to use Timer 2, but is impossible to set it from the OIL file like

COUNTER MyCounter {
  SOURCE = TIMER2_OVF;
  TICKSPERBASE = 6250;
  MAXALLOWEDVALUE = 6250;
  MINCYCLE = 1;
};

The idea was to use the entire 16 Mhz internal clock without no prescalar, in order to have a 16 us period between an overflow interrupt and the next one. Multiplying by 6250 I can have alarms associated to MyCounter of 100 ms or multiple of this value.

But when I try to compile the oil file using GOIL, I get an error saying that source can be only "Timer0_OVF, PIN1, PIN2". How I can use timer 2?

Another strange thing is that if i put source = timer0_ovf (like the default definition of SystemCounter), in the generated .c code I see an ISR(TIMER0_OVF). I think this is actually incorrect because the ISR associated with the timer0 overflow is TIMER0_OVF_vect, so it wouldn't work anyway. How SystemCounter works actually? 

Thank you

thegabriele97, 2020-11-10:
This is how my setup() function actually looks like: 

```C
void setup() {
#ifdef DEBUG
	Serial.begin(SERIAL_BAUDRATE);
#endif

	Serial.begin(SERIAL_BAUDRATE);

	// Setting timer0 to 0
	TCNT2 = 0x00;

	// Setting TOIE in order to enable interrupt
	TIMSK2 |= 0x1;

	// Clearing timer 2 configuration registers A and B
	TCCR2A &= 0x00;
	TCCR2B &= 0x00;

	// Setting scalar to 1: CS02, CS01, CS00 = 0b001
	TCCR2B |= 0x01;

	/*
	 * Now we have a Timer2 with a 16MHz clock. 
	 * Ath each clock cycle, TCNT0 goes up by 1.
	 * When it reaches 0xff, we have overflow and it
	 * means an interrupt: this means that each interrupt
	 * occurs every 16 us and in order to have a
	 * counter that counts 1 every 100 ms
	 * we need exactly 6250 ticks!
	 */

	pinMode(MORSE_LED, OUTPUT);
	digitalWrite(MORSE_LED, LOW);
}
```

mbriday, 2020-11-12:
Hi,
I have just pushed a fix. You can now use TIMER0 to 2.
I also added a full example (`` examples/avr/arduinoUno/customCounterExample/``). However, using other timers may have side effect (TIM2 is used for Tone and PWM!).
Regards

closed 2020-11-12 by mbriday