TrampolineRTOS/trampoline #98: goil Xcode project is missing project.pbxproj

andreivdev, 2021-04-25:
goil Cocoa app does not open any window on my M1 Mac so I wanted to run a debug build and check what's wrong.

I quickly got stuck because of this error

[image]

jlbirccyn, 2021-05-19:
I'll check that.

mbriday, 2021-05-20:
There was an update on commit 2624863293f67820a81ed5e7fda07480d1ade476. You may be able to compile and run on an M1 Mac (cli tool in goil/makefile-macosx and cocoa editor with the XCode project).

There are binaries provided (for OS X Silicon) and the links are on the main README.md. The binaries are **not** signed. You may self sign them or rebuild binaries from sources.

regards

closed 2021-05-20 by mbriday