TrampolineRTOS/trampoline #100: goil compilation error

wowscai, 2021-05-17:
When compiling goil (under MacOSX Big Sur 11.3.1), I have error during compiling. Below the output of build.py execution

makefile-macosx % ./build.py        
[  0%] Native Compiling for OS X: all-galgas.cpp
[  0%] Native Compiling for OS X: all-libpm.cpp
[  1%] Native Compiling for OS X: all-declarations-11.cpp
[  1%] Native Compiling for OS X: all-declarations-10.cpp
[  1%] Native Compiling for OS X: all-declarations-9.cpp
[  2%] Native Compiling for OS X: all-declarations-8.cpp
In file included from ../build/libpm/all-libpm.cpp:19:
[  2%] Native Compiling for OS X: all-declarations-7.cpp
../build/libpm/bdd/C_BDD-and-operation.cpp:38:15: error: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Werror,-Wnon-c-typedef-for-linkage]
typedef struct {
              ^
               tStructANDOperationCacheEntry
../build/libpm/bdd/C_BDD-and-operation.cpp:39:3: note: type is not C-compatible due to this member declaration
  public : uint64_t mOperands ;
  ^~~~~~~~
../build/libpm/bdd/C_BDD-and-operation.cpp:41:3: note: type is given name 'tStructANDOperationCacheEntry' for linkage purposes by this typedef declaration
} tStructANDOperationCacheEntry ;
  ^
1 error generated.
Return code: 1
Wait for job termination...
1 error.

Any ideas/solution? Thank you

jlbirccyn, 2021-05-19:
I'll check that. Unfortunately I do not have Big Sur yet.

mbriday, 2021-05-20:
Hi,
This have been fixed in commit c1d8582.

closed 2021-05-20 by mbriday