TrampolineRTOS/trampoline #107: Trampoline aarch64

grundmanns, 2022-05-09:
Hello Jean-Luc,
we ported our software package to Trampoline on a Cr7 core for demonstration purpose (to show that it runs with AUTOSAR).
We have in our SoC a CA53 core, where our Software should now be ported to.
We would like to show it with Trampoline AUTOSAR there.
I grep the Trampoline code we have but I cannot find support for aarch64 or arm64.
Is there any 64 bit arm support in Trampoline ?
We right now run it in FreeRTOS on the CA53 and it works but we want to have it on Trampoline too.

BR
Sven Grundmann

grundmanns, 2022-05-12:
Hello Jean-Luc,

thank you very much for your answer.
Right now we use our own machine port for our Cr7 core loosely based on the RPi armv7 bsp.
The question is if there is something for armv8.
Do you see major problem to port to armv8 ?
Sorry if I got your answer wrong and you are going to ask the RPi2 author exactly that.

BR
Sven Grundmann

jlbirccyn, 2022-05-12:
I do not see major problem but it will take some work. You will have to adapt the system call handler to use the 64 bits registers as long as the interrupt handler template. All this stuff is in assembly language. The tpl_sc_handler is commented extensively but if you have understanding problems, feel free ask :)

Best regards
--

grundmanns, 2022-05-12:
Hello Jean-Luc,
Ok I understand.
So I wait if [DavidGarriou](https://github.com/DavidGarriou) has something up to his sleeve, if not we have to think, if we port it ourselves but it is some major work.
I guess all that assembly stuff has to replaced. Does the C code have to be adapted too or is a simply recompile with aarch64 enough ?
BR
Sven Grundmann

jlbirccyn, 2022-05-12:
Probably not all. I am not very familiar with the aarch64 but in what I read from the white paper I downloaded from https://community.arm.com/arm-community-blogs/b/architectures-and-processors-blog/posts/porting-to-arm-64-bit the assembly languages are very similar. Maybe only a register name change is necessary but the ABI of aarch64 has to be checked too. 

I would say it is more documentation checking and some assembly instructions changes than major assembly language rewriting.

For C parts, it depends. Of course C code that deals with hardware init have to be adapted (everything that is in machines directory) but all stuff in os and autocar directory will compile without problem.

Best regards 
--
Jean-Luc Béchennec

grundmanns, 2022-05-16:
Hello Jean-Luc,

thank you for that information. I will let you know if I try that path.
For now I understand enough.

BR
Sven Grundmann

closed 2022-09-07 by grundmanns