TrampolineRTOS/trampoline #110: Build trampoline using cmake

grundmanns, 2022-07-08:
Hello Jean Luc,
we changed our build process of our software that is running on trampoline OS.
We are using cmake now.
The trampoline build process is based on oil file used to generate the python build files
build.py and make.py.
make.py is calling build.py as a child process.
I understood that depending on the settings in the oil file (e.g. oil file version scalability class selection) the build process is dynamic and modules are added and removed.

To map this dynamic changes in the build to our cmake, we have to change the templates for build.py / make.py to generate a CMakeLIsts file ?
Or do we have to create an own CMakeLists template that we enter somewhere and could activate with a oil file setting?

Do you have an idea what would be the best approach for that ?

BR
Sven Grundmann

mbriday, 2022-07-08:
Hi,
CMake support has been added a few months ago. 
In the .oil file, you have to set `SYSTEM=CMAKE`, instead of `SYSTEM=PYTHON`. There is also a VSCode integration for few boards (stm32): `SYSTEM=CMAKE{VSCODE=TRUE;};`
The `CMakelists.txt` is generated from a [goil template](https://github.com/TrampolineRTOS/trampoline/blob/master/goil/templates/build/cmakelists_txt.goilTemplate)
You can have a look to the [updated documentation](https://github.com/TrampolineRTOS/trampoline/blob/master/documentation/manual/main.pdf), chapter 15.
regards,
Mik

grundmanns, 2022-07-08:
Hello Mik,
thank you very much for the quick answer.
This is very good news.
I will report it to my team.
BR
Sven Grundmann

closed 2022-07-08 by grundmanns