TrampolineRTOS/trampoline #116: arm-none-eabi-ld: warning: blink_exe has a LOAD segment with RWX permissions

peter-samir-luis, 2023-04-11:
i am trying to compile this example "examples\cortex\armv7em\stm32f407\stm32f4discovery\blink"
i run: 
1. "goil --target=cortex/armv7em/stm32f407/stm32f4discovery --templates=../../../../../../goil/templates/ blink.oil"
2. "python make.py"

this is the last message printed on the screen 
" arm-none-eabi-ld: warning: blink_exe has a LOAD segment with RWX permissions
←[1m←[91mReturn code: 1←[0m
←[1m←[91m1 error.←[0m "

 blink_exe.bin file is NOT generated .

jlbirccyn, 2023-04-11:
Hello,

Just removed --fatal-warnings ld option as a temporary workaround. You can pull

Best regards

jlbirccyn, 2023-04-11:
Fixed with 02c4110e33d18bfd51dfa88af878c5650a3073f5

closed 2023-04-11 by jlbirccyn