TrampolineRTOS/trampoline #117: examples

peter-samir-luis, 2023-04-12:
Could you help me please?
I want to extend the instruction set of riscv processor to speed up the execution of operating systems.

I have chosen to do analysis on trampoline with arm cortex processors or PowerPc.
I will use the disassembly file to figure out the most frequently assembly instructions, then add it to riscv instruction set if it does not exist.

I need examples or projects that use most features of trampoline.

Do you suggest another way to make this analysis?

As you participated in trampoline, do you have an idea for instructions that you think will speed up the execution?

Thank you in advance.

jlbirccyn, 2023-04-12:
Hello,

In what context do you want to do this? Are you a master student or something like that?

I'm not sure if this method gives relevant results. All 3 processors are RISC and have semantically similar instruction sets. I'm also not sure that adding a few instructions will significantly speed up the OS.

I suggest that you do some bibliography on the subject:

https://scholar.google.com/scholar?hl=fr&as_sdt=0%2C5&q=OS+acceleration+instruction+set&btnG=

https://scholar.google.com/scholar?hl=fr&as_sdt=0%2C5&q=OS+hardware+acceleration+instruction+set&btnG=

Best regards

jlbirccyn, 2023-04-12:
The OS code is compiled according to the kind of objects declared in the OIL file. So, to have the whole code compiled, you need to supply an OIL file with at least one object of each kind : 1 task, 1 ISR, 1 event, 1 alarm an so on. You do not need to provide a functioning application and you do not need to provide many applications, one is enough.

peter-samir-luis, 2023-04-12:
I have not worked with oil before .so I need to learn how to write an oil file and this is out of the scope of the project and will take time and I am limited by deadlines. It would be a great favor if you could write the oil for me.

jlbirccyn, 2023-04-13:
Obviously this is something which is out of scope of my current projects, will take time and I am limited by deadlines too.

You have a lot of OIL files in examples/ and tests/ directory.