TrampolineRTOS/trampoline #118: Trampoline with IDE (KeilC/IAR)

quyleduc, 2023-04-12:
Currently, I want to add the generated files and source file of Trampoline to the IDE which supports STM32 to debug and trace. How can I do that?

Koryphon, 2023-04-12:
Hello,

I have no idea how to add generated files to an IDE like KeilC/IAR. However, I assume that the IDE has a project file describing how the project is structured. If the specifications are open, it should be possible to generate this file from the OIL via a goil template.

Sincerely

jlbirccyn, 2023-04-12:
Hello,

I have no idea how to add generated files to an IDE like KeilC/IAR. However, I assume that the IDE has a project file describing how the project is structured. If the specifications are open, it should be possible to generate this file from the OIL via a goil template.

Sincerely

quyleduc, 2023-04-12:
Hi,

Thank you for your response. yes, you are right, but I need to base it on the goil template and it is too difficult when I change the feature.
Do you know how to debug code instead of using command line debug?

jlbirccyn, 2023-04-12:
I assign the issue to my colleague who did that with VSCode.

mbriday, 2023-04-13:
Hi,
The goil OIL compiler provided with Trampoline generates a set of build rules the first time it is run. These build rules may be generated as a python script (in most of the examples provided in the example/ directory), or as a CMake compliant project.

If your IDE is able to deal with CMake (such as VSCode or Qt Creator) it will be integrated in the IDE easily. You can have a look at the [Trampoline doc, section 15.2](https://github.com/TrampolineRTOS/trampoline/blob/master/documentation/manual/main.pdf). The only drawback of using CMake is that if you update the `.oil` file, you will have to compile your app twice (the first run goil and updates sources, the second will compile).

There is an howto  `Trampoline with VSCode` [here](https://gitlab.univ-nantes.fr/briday-m/itii-mac-tp-etu/-/blob/master/doc/vscode.md) with the related tools (cross-compiler, …) [here](https://gitlab.univ-nantes.fr/briday-m/itii-mac-tp-etu/-/blob/master/doc/tools.md). The target is a STM32 Cortex, but it can be adapted to other cortex targets.

quyleduc, 2023-04-13:
Yeah. That's the very thing i need. Thank you for your information

closed 2023-04-13 by mbriday