TrampolineRTOS/trampoline #120: [Question] Does the context switch routine have consider "Pendsv" handler?

Nigelwz, 2023-04-14:
- In some os, like freertos, it can run PendSV() in context switch.. Does the osek or autosar have consider PendSv check?

jlbirccyn, 2023-04-14:
Hello,

You mean that the context switching is done by the PendSV handler I guess?

Best regards

jlbirccyn, 2023-04-14:
I still don't understand what you are saying. If the context switch occurs in the Systick or timer handler then it does not occur in the PendSV handler.

If you mean that the context switch occurs in the PendSV handler and the PendSV was triggered by the Systick or Timer handler then I don't see the point of invoking a PendSV rather than doing the context switch directly at the end of the Systick or Timer handler.

Also, I doubt that context switching is only done asynchronously in the handler of a hardware IT. When a synchronous system call leads to a context switch, it should be done immediately, not 1ms later (at worst).

Nigelwz, 2023-04-14:
Thanks for your information!!  
That's my first time to investigate the automotive indudtry.
Because **sometime** the "systick interrupt" and "hardware interrupt" can occur simultaneously , (e.g: uart, spi, button, hardware fault...), so some os can set the systick's priority is lower than other hardware interrupt, then if the systick trigger, it can enable the pendsv interrupt, then do context switch in pendsv.
Below is sample code..
```
void xPortSysTickHandler( void )
{
	/* The SysTick runs at the lowest interrupt priority, so when this interrupt
	executes all interrupts must be unmasked.  There is therefore no need to
	save and then restore the interrupt mask value as its value is already
	known.
	 */
	portDISABLE_INTERRUPTS();
	{
		/* 
		Increment the RTOS tick. 
		*/
		if( xTaskIncrementTick() != pdFALSE )
		{
			/* 
			A context switch is required.  Context switching is performed in
			the PendSV interrupt.  Pend the PendSV interrupt. 
			*/
			portNVIC_INT_CTRL_REG = portNVIC_PENDSVSET_BIT;
		}
	}
	/*
	Enable interrupt, increntment pensv ISR.
	*/
	portENABLE_INTERRUPTS();
}
BaseType_t xTaskIncrementTick( void )
{
/// do context switch....
}
`

Or we just need to set ISR1 or ISR2 on trampline, is it correct?

closed 2023-10-14 by jlbirccyn