TrampolineRTOS/trampoline #121: Le cas de communication samd21 et raspberry ne fonctionne pas

moustafa-x, 2023-04-18:
Pouvez-vous aider à résoudre un problème !
#include "tpl_os.h"
//the READY tasks state definition conflicts with registers definition (in sam.h)
#undef READY
#include "sam.h"
#include <pinAccess.h>
#include "board/iutNantes/include/timer.h"
#include "board/iutNantes/include/oled.h"
#include <eic.h>
#include "serial.h"
#include <errno.h>
#include <stdio.h> //printf
//distance en cm.
volatile unsigned int range;

int main(void)
{
	pinMode(PORTB,9,OUTPUT);	//TRIGGER pin
	pinMode(PORTB,15,INPUT);	//ECHO pin
	EICInitClock(F1MHZ);
	TCinitClock(F1MHZ,3); //1 pulse/us
	OLEDInit(0);
     SERIALInit(115200);
 	range = 0;
	StartOS(OSDEFAULTAPPMODE);
	return 0;
}

TASK(displayTask)
{
	OLEDSetLine(0);
	OLEDPrintString("distance ");
	OLEDPrintInt(range,3);
	OLEDPrintString("cm.");
	TerminateTask();
}

TASK(timeoutTask)
{
	OLEDSetLine(0);
	OLEDPrintInt(range,3);
	OLEDPrintString("  Pas de capteur!");
	TerminateTask();
}

TASK(measureTask)
{
	//trigger
	digitalWrite(PORTB,9,1);
	for(volatile int i=0;i<1000;i++);
	digitalWrite(PORTB,9,0);
	
	//lancement alarme du timeout
	SetRelAlarm(activateTimeout,35,0);
	TerminateTask();
}

ISR(triggerISR)
{
	if(digitalRead(PORTB,15) == 1)
	{
		//front montant. Lancement timer TC3.
		TC3->COUNT16.COUNT.reg = 0; //init val
		TC3->COUNT16.CTRLA.reg = 2; //run, sans prescaler.
	} else {
		unsigned int mes = TC3->COUNT16.COUNT.reg; //1 tick/us
		range = mes/58; //conversion en cm.
		TC3->COUNT16.CTRLA.reg = 0; //arret (non necessaire)
		 
	}
	CancelAlarm(activateTimeout);
}
typedef struct {
    uint16_t id; // ID du capteur
    uint16_t value; // Valeur du capteur
} SensorData;
TASK(STask){
	        SensorData sensorData;
		    sensorData.id = 1; // ID du capteur
            sensorData.value = range; // Valeur du capteur
		    char s[5]={'s','s','s','s','s'};
		SERIALPutString(s); // Envoi du message initial
	    for (int i = 0; i < sizeof(sensorData); i++) {
        SERIALPutchar(*((uint8_t*)&sensorData + i));}
		
 TerminateTask();
}

mbriday, 2023-04-18:
Bonjour,

Quel problème?

moustafa-x, 2023-04-18:
Aucune donnée n'est envoyée. Je ne connaissais pas de solution.

moustafa-x, 2023-04-18:
[image]
  aucune donnée n'a été envoyée

jlbirccyn, 2023-04-18:
Quelle est la description OIL de la tâche STask ?

moustafa-x, 2023-04-18:
OIL_VERSION = "2.5";

IMPLEMENTATION trampoline {
  /* This fix the default STACKSIZE of tasks */
  TASK {
    UINT32 STACKSIZE = 800 ;
  };
  /* This fix the default STACKSIZE of ISRs */
  ISR {
    UINT32 STACKSIZE = 500 ;
  } ;
};

CPU blink {
  OS config {
    STATUS = EXTENDED;
    
    BUILD = TRUE {
      TRAMPOLINE_BASE_PATH = "../../../../../../";	
      APP_SRC = "board/iutNantes/src/adc.c";
      APP_SRC = "board/iutNantes/src/oled.c";
      APP_SRC = "board/iutNantes/src/serial.c";
      APP_SRC = "board/iutNantes/src/dac.c";
      APP_SRC = "board/iutNantes/src/redirectPrintf.c";
      APP_SRC = "board/iutNantes/src/timer.c";
      CFLAGS  = "-I board/iutNantes/include"; 
      APP_SRC = "blink.c";
      APP_NAME = "blink.elf";
      CFLAGS  = "-O3 -std=c99"; 
      COMPILER = "arm-none-eabi-gcc";
      ASSEMBLER = "arm-none-eabi-as";
      LINKER = "arm-none-eabi-ld";
      COPIER = "arm-none-eabi-objcopy";
      SYSTEM = PYTHON;
    };
    
    SYSTEM_CALL = TRUE;
    MEMMAP = TRUE {
      COMPILER = gcc;
      LINKER = gnu_ld { SCRIPT = "script.ld"; };
      ASSEMBLER = gnu_as;
      MEMORY_PROTECTION = FALSE;
    };
  };
  
  APPMODE std {};
  
  TASK measureTask {
    PRIORITY = 5;
    AUTOSTART = FALSE;
    ACTIVATION = 1;
    SCHEDULE = FULL;
  };
  
  ALARM activatePeriodic {
    COUNTER = SystemCounter;
    ACTION = ACTIVATETASK {
      TASK = measureTask;
    };
    AUTOSTART = TRUE {
      APPMODE = std;
      ALARMTIME = 100;
      CYCLETIME = 100;
    };
  };
  
  TASK timeoutTask {
    PRIORITY = 3;
    AUTOSTART = FALSE;
    ACTIVATION = 1;
    SCHEDULE = FULL;
  };
  
  ALARM activateTimeout {
    COUNTER = SystemCounter;
    ACTION = ACTIVATETASK {
      TASK = timeoutTask;
    };
    AUTOSTART = FALSE;
  };

  TASK displayTask {
    PRIORITY = 1;
    AUTOSTART = FALSE;
    ACTIVATION = 1;
    SCHEDULE = FULL;
  };
  
  ALARM activateDisplay {
    COUNTER = SystemCounter;
    ACTION = ACTIVATETASK {
      TASK = displayTask;
    };
    AUTOSTART = TRUE {
      APPMODE = std;
      ALARMTIME = 1000;
      CYCLETIME = 1000;
    };
  };

 TASK STask {
    PRIORITY = 4;
    AUTOSTART = FALSE;
    ACTIVATION = 1;
    SCHEDULE = FULL;
  };
  
  ALARM comTASK {
    COUNTER = SystemCounter;
    ACTION = ACTIVATETASK {
      TASK = STask;
    };
    AUTOSTART = TRUE {
      APPMODE = std;
      ALARMTIME = 1000;
      CYCLETIME = 1000;
    };
  };

  ISR triggerISR {
    CATEGORY = 2;
    PRIORITY = 30;
    STACKSIZE = 256;
    SOURCE = EIC_IRQ {
      PIN = PB15 {
        TRIGGER = BOTH;
        PULL = NONE;
        FILTERING = FALSE;
      };
    };
  };
  

  
};

moustafa-x, 2023-04-18:
Je vous ai envoyé un e-mail

jlbirccyn, 2023-04-18:
Ajoutez un affichage sur l'OLED dans Task pour savoir si cette tâche s'exécute.

moustafa-x, 2023-04-18:
pour le moment je ne l'ai pas .
Y a-t-il un autre moyen ?

jlbirccyn, 2023-04-18:
Je n'ai pas compris votre réponse. Vous n'avez pas quoi ?

moustafa-x, 2023-04-18:
afficheur OLED

jlbirccyn, 2023-04-18:
Dans ce cas, changez l'état d'une des LED de la carte.

moustafa-x, 2023-04-18:
Même problème, pouvez-vous voir si le code est correct ? 
Je vous ai envoyé un e-mail tout les fichiers que j'ai utilisé.

jlbirccyn, 2023-04-18:
Je ne reçois pas vos email (je ne sais pas pourquoi). Vous pouvez attacher des fichiers ici.

moustafa-x, 2023-04-18:
[f.zip](https://github.com/TrampolineRTOS/trampoline/files/11265869/f.zip)

moustafa-x, 2023-04-18:
C'est tout fichier avec toutes les librairies

jlbirccyn, 2023-04-18:
Vous n'avez pas la carte sous la main ?

moustafa-x, 2023-04-18:
la carte sous la main mais problème de fonctionnement de task serial ,je sais pas est ce qu le code correct ou non?

moustafa-x, 2023-04-18:
J'ai ajouté led mais  ne fonctionne pas

jlbirccyn, 2023-04-19:
1) La LED clignote si vous mettez le code de changement de son état dans displayTask ?
2) Si oui, qui se passe-t-il si vous faite clignotez la LED dans STask et que vous commentez le reste du code de STask (ie STask ne fait que clignoter la LED et rien d'autre.

moustafa-x, 2023-04-27:
Après de nombreuses tentatives, je n'ai pas pu envoyer les données. Quelqu'un peut-il aider ? (Je pense qu'il s'agit d'un travail pratique effectué par vos étudiants)

moustafa-x, 2023-05-01:
J'ai un problème avec STack 
Comment trouver une solution?
Aussi j'ai aussi essayé avec arduino, j'ai eu le même problème, une seule tâche fonctionne pour moi

moustafa-x, 2023-05-03:
/ #
[githab.zip](https://github.com/TrampolineRTOS/trampoline/files/11381911/githab.zip)

Essalamou Alaikom .
j'espère que vous allez bien?.

// j'ai fait dans tout les task une led au fonctionne normal
// jai réussi la trasmision de data mais le cas
(SERIALPutchar(((uint8_t*)&sensorData + i));) ne renvoie rien
// cette intrpation(triggerISR) ne fonctionne pas.

Cordialement.

closed 2023-10-13 by jlbirccyn