TrampolineRTOS/trampoline #127: Unbounded write by strcpy

szsam, 2023-06-16:
https://github.com/TrampolineRTOS/trampoline/blob/7ac178b0f67268735e8f9084cac49efe545ae80c/machines/posix/tpl_viper_interface.c#L83
This 'call to strcpy' with input from call to getenv may overflow the destination.

sefau, 2023-10-26:
We should book room to append "/viper" at the end of the string.

Machiry, 2024-03-25:
Hello @sefau ,

My name is Aravind Machiry, Assistant Professor at Purdue's ECE Department.

Thank you for considering this pull request. This pull request was the result of our on-going research work (along with @szsam) to improve the security and quality of open-source embedded projects.

In addition to scanning codebases with CodeQL, we are also doing a short (~4 minutes) survey to understand the use of static analysis tools like `gcc -Wall` and CodeQL in embedded software projects.

It would greatly benefit our research if you could fill this anonymous survey: https://purdue.ca1.qualtrics.com/jfe/form/SV_0OnXfr5plPe1QCa

Thank you,
Aravind

commit 384c893d (2023-10-10): Fix #127
commit c2583e2a (2023-10-26): Fix #127