TrampolineRTOS/trampoline #138: ARM Cortex directories naming

RICCIARDI-Adrien, 2023-10-10:
Hi,

In the Trampoline machines directory, there is the `cortex` directory that seems to be dedicated to microcontroller Cortex-M versions, then there is the `cortex-a` directory for the Cortex-A7, and there is also the `arm` directory for the pre-Cortex architectures, like ARM9.

Maybe it would  make more sense to rename the `cortex` directory to `cortex-m`, so it is easier understand ?

A second question : if someone wants to add a Cortex-R port, are you OK to create a dedicated `cortex-r` directory or do you prefer to share an existing directory ?

Koryphon, 2023-10-10:
Hello Adrien,

cortex could be rename cortex-m provided templates are changed accordingly.

For Cortex-R (that I know little about) It depends if the sc_handler and interrupt routine code are (or would be) different or not. I see that the -R supports both ARM and thumb, and I assume cross-calls. Therefore, it depends on whether the sc_handler is written in thumb or ARM. To be discussed.

jlbirccyn, 2023-10-10:
Hello Adrien,

cortex could be rename cortex-m provided templates are changed accordingly.

For Cortex-R (that I know little about) It depends if the sc_handler and interrupt routine code are (or would be) different or not. I see that the -R supports both ARM and thumb, and I assume cross-calls. Therefore, it depends on whether the sc_handler is written in thumb or ARM. To be discussed.

RICCIARDI-Adrien, 2023-10-10:
Hello Jean-Luc,

Thanks for the quick reply.

What is the `simple` machine ? It seems to provide the same Cortex-M architectures than the current `cortex` directory ?

jlbirccyn, 2023-10-10:
```simple``` was used in a lab where students implemented a simple context switch (if you look, the sc_handler is missing). As of last year, the lab has been ported to the Raspberry Pi Pico and ``simple`` is no longer used, nor will it be in the future. It will therefore be removed. ``simple`` has not been put on a branch to make it easier for students who are not very familiar with Git to set up the lab.

RICCIARDI-Adrien, 2023-10-10:
OK, do you want to remove the `simple` machine and examples yourself or do you prefer a pull request ?
And same question for the `cortex` directories, do you want to rename them or do you prefer a pull request ?

RICCIARDI-Adrien, 2023-10-10:
I have now the answer for the `simple` architecture !
Are the following directories also part of the `simple` architecture ?
* `examples/simple/`
* `goil/templates/linker/gnu_ld/simple/`
* `goil/templates/config/simple/`

jlbirccyn, 2023-10-10:
Yes,

I just pushed the removing of these directory

For the name change of cortex to cortex-m, I welcome a pull-request.

RICCIARDI-Adrien, 2023-10-10:
Got it for the `cortex-m`.

Do you want to keep the `examples/simple/` directory ?

RICCIARDI-Adrien, 2023-10-25:
The pull request renaming `cortex` into `cortex-m` is here : https://github.com/TrampolineRTOS/trampoline/pull/145.

RICCIARDI-Adrien, 2023-11-06:
The pull request renaming `cortex-a` into `cortex-a-r` is here : https://github.com/TrampolineRTOS/trampoline/pull/147.

closed 2023-11-06 by RICCIARDI-Adrien