TrampolineRTOS/trampoline #142: Example build issue : Teensy 3.1 isr1onFTM

RICCIARDI-Adrien, 2023-10-25:
```
ar@t490s:~/Documents/Renesas/trampoline_fork/examples/cortex/armv7em/mk20dx256/teensy31/isr1onFTM$ ../../../../../../goil/makefile-unix/goil --target=cortex/armv7em/mk20dx256/teensy31 --templates=../../../../../../goil/templates isr1onFTM.oil 
Replaced 'isr1onFTM/tpl_dispatch_table.c'.
Replaced 'isr1onFTM/tpl_invoque.S'.
Replaced 'isr1onFTM/tpl_os.h'.
Replaced 'isr1onFTM/tpl_service_ids.h'.
Replaced 'build.py'.
Replaced 'make.py'.
Replaced 'isr1onFTM/tpl_app_custom_types.h'.
Replaced 'isr1onFTM/tpl_app_config.c'.
Replaced 'isr1onFTM/tpl_app_config.h'.
Replaced 'isr1onFTM/tpl_app_define.h'.
Replaced 'isr1onFTM/MemMap.h'.
Replaced 'isr1onFTM/Compiler.h'.
Replaced 'isr1onFTM/Compiler_Cfg.h'.
Replaced 'isr1onFTM/script.ld'.
Replaced 'isr1onFTM/AsMemMap.h'.
Replaced 'isr1onFTM/tpl_static_info.json'.
Replaced 'isr1onFTM/tpl_vectors.c'.
Replaced 'isr1onFTM/tpl_primary_irq.S'.
Replaced 'isr1onFTM/cmsis_wrapper.h'.
Replaced 'isr1onFTM/tpl_external_interrupts.c'.
Replaced 'isr1onFTM/tpl_app_interrupts.c'.
/home/ar/Documents/Renesas/trampoline_fork/examples/cortex/armv7em/mk20dx256/teensy31/isr1onFTM/isr1onFTM.oil:129:14:21:
semantic error #1: The priority of this interrupt source is not settable
    SOURCE = FTM0_IRQ;
-------------^^^^^^^^

/home/ar/Documents/Renesas/trampoline_fork/goil/templates/code/cortex/armv7em/mk20dx256/teensy31/custom_code.goilTemplate:130:2:9:
semantic error #2: runtime error
 template if exists tpl_cortex_definitions_h in code
-^^^^^^^^

/home/ar/Documents/Renesas/trampoline_fork/goil/templates/code/cortex/armv7em/mk20dx256/teensy31/custom_code.goilTemplate:129:1:5:
semantic error #3: runtime error
write to PROJECT+"/tpl_cortex_definitions.h":
^^^^^

/home/ar/Documents/Renesas/trampoline_fork/goil/templates/root.goilTemplate:1554:1:8:
semantic error #4: runtime error
template if exists custom_code in code
^^^^^^^^

Replaced '/home/ar/Documents/Renesas/trampoline_fork/examples/cortex/armv7em/mk20dx256/teensy31/isr1onFTM/build/isr1onFTM.oil.dep'.
No warning, 4 errors.
ar@t490s:~/Documents/Renesas/trampoline_fork/examples/cortex/armv7em/mk20dx256/teensy31/isr1onFTM
```

jlbirccyn, 2023-10-25:
Yes. The Teensy port is broken and has been for quite some time. I won't have time to deal with this for a long time. It's low priority.

I suggest removing the Teensy port from the workflow for now.

RICCIARDI-Adrien, 2023-10-25:
OK, let's postpone this. Do I close the Teensy-related tickets ?

jlbirccyn, 2023-10-25:
No, let them. Otherwise, we risk forgetting that this port is to be revised. 😄

jlbirccyn, 2023-11-08:
See #104

closed 2023-11-08 by jlbirccyn