TrampolineRTOS/trampoline #163: tpl_invoque.s goes hardfault

HassanMH01, 2024-03-17:
I'm porting trampoline os to tiva c target "tm4c123gh6pm", have written required files for porting and now debugging, processor goes into hard fault after this assembly line which can be found in tpl_invoque.s in line 108:

    /*-------------------------------------------------------------------------
     * return from the service wrapper
     */
    bx   lr
StartOS_direct_call:

any information to help?

mbriday, 2024-03-17:
Hi,
It seems that you have broken the stack:
 * Is the stack size sufficient?: `STACKSIZE` parameter of the task.
 * maybe you should track the stack usage through the svc handler.

 You can init the stack zone during task init, with the `PAINT_STACK` parameter (in `CPU->OS`):
```
CPU blink {
  OS config {
    PAINT_STACK = TRUE;
```
(see `tpl_init_context` in `tpl_machine_cortex.c`).
Regards
Mik

HassanMH01, 2024-03-18:
yes, it's in my implementation and still have this error
problem starts from here in tpl_os_os_kernel.c:
`IF_NO_EXTENDED_ERROR(result)
  {
#if NUMBER_OF_CORES > 1
    /*
     * Sync barrier at start of tpl_start_os_service.
     */
    tpl_sync_barrier(&tpl_start_count_0, &tpl_startos_sync_lock);

    application_mode[core_id] = mode;
#else
    application_mode = mode;
#endif

    tpl_init_os(mode);

    tpl_enable_counters();

    /*
     * Call the startup hook. According to the spec, it should be called
     * after the os is initialized and before the scheduler is running
     */
    CALL_STARTUP_HOOK()

    /*
     * Call the OS Application startup hooks if needed
     */
    CALL_OSAPPLICATION_STARTUP_HOOKS()

#if NUMBER_OF_CORES > 1
    /*
     * Sync barrier just before starting the scheduling.
     */
    tpl_sync_barrier(&tpl_start_count_1, &tpl_startos_sync_lock);
#endif

    /*
     * Call tpl_start_scheduling to elect the highest priority task
     * if such a task exists.
     */
    tpl_start_scheduling(CORE_ID_OR_NOTHING(core_id));

    SWITCH_CONTEXT_NOSAVE(core_id)
  }

  PROCESS_ERROR(result)

  /*  unlock the kernel  */
  UNLOCK_KERNEL()
}`
as 
`result = 0x07`

mbriday, 2024-03-18:
It doesn't seem like a question :/

HassanMH01, 2024-03-18:
I don't know how to resolve such a bug

jlbirccyn, 2024-03-18:
Hello.

If you get ```result == 7``` (```7``` means ```E_OS_STATE```) in ```tpl_start_os_service```, it means you are calling ```StartOS``` a second time from a task. If you do not do that then it means initialized global/static variables are not initialized correctly.

```E_OS_STATE``` is returned if the current state of the OS is not ```OS_INIT```. Trace execution within ```tpl_current_os_state``` to see what state is returned.

HassanMH01, 2024-04-15:
Ok, I've passed this error and now I encounter a situation in tpl_os_timeobj_kernel with current time object 

[image]
[image]
[image]

As it's not a valid location it goes hardfault

mbriday, 2024-04-16:
Hi,
Have you updated the link script for your specific target? It seems that there is a problem with memory locations.
regards,

HassanMH01, 2024-04-16:
Yes goil template files are updated with
Flash, RAM and stack size

mbriday, 2024-04-16:
It seems that time_obj does not refer to a correct memory location. Valid RAM is most of the time 0x20xxxxxx, but pointers are in 0x0100xxxx… Maybe you have to track the time objs? and that the double linked list is correct.
regards