TrampolineRTOS/trampoline #164: Porting to SMT32G0

casas45, 2024-03-25:
Hello.
I want to port the OS to a STM32G0 mcu (core board), which has the Cortex m0+, can someone tell me which files I need to modify, considering the CPU is already supported.
I will really appreciate your help.

mbriday, 2024-03-25:
Hi,
We're in the process of drafting documentation for STM32 boards. Even if the document was written with Cortex M4/M7 in mind, it should be able to serve as a basis for your porting. This draft uses a python script that has not yet been put on master, but is available in the `stm32h743` branch (porting in progress). 

 # Goil templates
 The templates should be adapted in 4 parts:
  * `config` : OIL configuration, including interrupts, libraries, …
  * `code`   :
  * `linker` : ld script (memory mapping)
  * `build`  : build rules (cmake specific, …)

 ## template/config

 In config templates, each file inherits from higher level directories. For instance, the `config.oil` is built from:
 * `goil/templates/config/config.oil` : common part
 * `goil/templates/config/cortex-m/config.oil` : common to cortex-m architecture
 * `goil/templates/config/cortex-m/armv7em/config.oil` : common to cortex M4/M7, Thumb2 based ISA
 * `goil/templates/config/cortex-m/armv7em/stm32h743/config.oil` : specific to the STM32H743

 First, the config templates should be adapted. Copy the configuration template from another port (i.e. `STM32L432`) to the new arch (i.e. `STM32H743`): `goil/templates/config/cortex-m/armv7em/stm32h743`:

Files are:
 * `config.oil`: main file with configuration: build options, files to include, paths
 * `interruptDefs.oil` may be generated by `setInterruptDefsOil.py` python script, thanks to the ST cmsis header file.
 * `interruptSources.oil` may be **partially** generated by the same script. It takes only into account specificities for EXTI and U(S)ART

 ## template/code

Most of these files are generic to the STM32. They are related to the way the interrupts are handled (exti for instance).

You have to update the `tpl_wrapper.goilTemplate` to the name of the main register header file.

 ## template/linker

The linker scripts are pretty much the same for a given family (CortexM4, cortexM7,...), except for:
 * the memory size which is defined in the `memory_map.goilTemplate` file.
 * the `_estack` symbol in the beginning of the main template `script.goilTemplate`

 ## template/build

 The build scripts have been already defined in the config templates. However, one can set VSCode related tools (when using the CMake build system) for a better integration. The directory should be copied from another port (stm32l432 for instance) and `vscode_launch_json.goilTemplate` should be adapted with
  * the `device`, used when debugging by ST-Link
  * the SVD file, used by VSCode ARM extension to deal with peripheral registers.

# Code
Some files should be specialized in the `machine` directory

## machine
We need to take a file from a similar target (CM4/CM7). In fact, only one file should be updated: `system_stm32xx.c`, but this can be extracted from ST Cube IDE.

Different files (copied from the STM32L432 target):

 * `system_stm32xx.c`: contains the SystemInit() function, that largely depends on the clock tree. This part is specific! However, if you can build an example from ST Cube IDE, you can get the `system_stm32hxxx.c` file direcly. 
 * `tpl_trace.c`: allows you to trace kernel events (debug). It depends on uart. No other dependencies. Not compiled if the TRACE is not enabled in the .oil file.
 * `startup_stmxx.c`: second stage startup (in C): init .bss, .data, libc. This part is linked with symbols defined in the linker script. It should be similar for most STM32 flavors. Note: first stage startup is common to each cortex-m (in `tpl_startup.S`)
 * `tpl_machine_stmxx.c`: contains the Systick configuration. Similar to most targets, as the Systick is provided by ARM.
 * `handlers_stm32...c/.h`: these are ARM related handlers. They should not be different from one target to another.
 * `tpl_memory_protection.c/.h`: MPU config. Not required to make Trampoline start…

## example
To build the first application, a basic application should be defined. The best way is to copy a basic app (blink) from another target, and get .c, .oil and README files.

casas45, 2024-04-05:
I got it. Thanks for your help.

mbriday, 2024-04-08:
Great,
Could you make a pull request to integrate the port into the master branch?
regards,
Mik

rafal19921989, 2024-04-30:
Hi @mbriday 

Where can I find or when will the script you mentioned above - setInterruptDefsOil.py - be available?

mbriday, 2024-04-30:
Hi,
It is available in the `stm32h743` branch [here](https://github.com/TrampolineRTOS/trampoline/blob/stm32h743/goil/templates/config/cortex-m/armv7em/setInterruptDefsOil.py).
This is just an helper to port STM32 cortex-m targets. There are still some few parameters that are hard-coded.
regards,
Mik

rafal19921989, 2024-05-16:
@mbriday thanks for your reply.

I have a question about external communication - between ECUs.
Is it supported by Trampoline?